v4l2loopback/v4l2loopback #78: Cannot see the v4l2loopback using google-chrome or chromium

thierryg, 2015-01-16:
Hi, 

While the v4l2loopback is loaded i can play normaly with using ffmpeg/ffplay/vlc/... but when i start the google-chrome browser (or even chromium), i can only see my physical local webcam not my loopbak one.

I have see the following issue : https://code.google.com/p/chromium/issues/detail?id=142568

It explain that chomium filter the loopback webcam from the v4l2 device listing.

Is there someone that have made it works in this case?

Thanks for any reply but i haven't seen any previous issue on your bugtracker.

Regards

Thierry GAYET

umlaeute, 2015-01-16:
since `0.7.0` (hey, 0.8.0 was released more than a year ago!!) the driver supports an `exclusive_caps` option that makes it compatible with chrome (as is documented in the NEWS file).

enable it:

    # insmod v4l2loopback exclusive_caps=1

the newly created device will announce *output* capabilities only (so ordinary webcam applications (including chrome) won't see it. as soon as you have attached a producer to the device, it will start announcing *capture* capabilities only, so *chrome* (and other ~~braindead~~ applications) will see it)

umlaeute, 2015-01-16:
btw, the issue is a duplicate of #40

thierryg, 2015-01-17:
Hi, thanks for your previous reply. I have seen the bug #40 but it was not working for me.

Here is all the process i have followed for my test ; tell me is something is missing :

1. Build :

make clean && make

2. Check parameters :

modinfo v4l2loopback 
filename:       /lib/modules/3.13.0-37-generic/extra/v4l2loopback.ko
license:        GPL
author:         Vasily Levin, IOhannes m zmoelnig <zmoelnig@iem.at>,Stefan Diewald,Anton Novikov
description:    V4L2 loopback video device
srcversion:     DD5EF63321657CC1C01E35A
depends:        videodev
vermagic:       3.13.0-37-generic SMP mod_unload modversions 
parm:           debug:debugging level (higher values == more verbose) (int)
parm:           max_buffers:how many buffers should be allocated (int)
parm:           max_openers:how many users can open loopback device (int)
parm:           devices:how many devices should be created (int)
parm:           video_nr:video device numbers (-1=auto, 0=/dev/video0, etc.) (array of int)
parm:           card_label:card labels for every device (array of charp)
parm:           exclusive_caps:whether to announce OUTPUT/CAPTURE capabilities exclusively or not (array of bool)
parm:           max_width:maximum frame width (int)
parm:           max_height:maximum frame height (int)

3. Unload previous :

rmmod -r v4l2loopback 
or 
modprove -r v4l2loopback

4. Load the current drivers :

sudo su
modprobe videodev
insmod ./v4l2loopback.ko devices=1 video_nr=2 exclusive_caps=1

5. Check configuration

tree /sys/module/v4l2loopback/ 
/sys/module/v4l2loopback/
├── coresize
├── holders
├── initsize
├── initstate
├── notes
├── parameters
│   ├── debug
│   ├── exclusive_caps
│   ├── max_buffers
│   ├── max_height
│   ├── max_openers
│   ├── max_width
│   └── video_nr
├── refcnt
├── sections
│   ├── __bug_table
│   ├── __mcount_loc
│   └── __param
├── srcversion
├── taint
└── uevent

4 directories, 17 files

# cat /sys/module/v4l2loopback/parameters/exclusive_caps 
Y,Y,Y,Y,Y,Y,Y,Y

# ls -al /dev/video*
crw-rw----+ 1 root video 81, 0 janv. 17 15:59 /dev/video0

# ffmpeg -f v4l2 -list_formats all -i /dev/video0
ffmpeg version 1.2.6-7:1.2.6-1~trusty1 Copyright (c) 2000-2014 the FFmpeg developers
  built on Apr 26 2014 18:52:58 with gcc 4.8 (Ubuntu 4.8.2-19ubuntu1)
  configuration: --arch=amd64 --disable-stripping --enable-avresample --enable-pthreads --enable-runtime-cpudetect --extra-version='7:1.2.6-1~trusty1' --libdir=/usr/lib/x86_64-linux-gnu --prefix=/usr --enable-bzlib --enable-libdc1394 --enable-libfreetype --enable-frei0r --enable-gnutls --enable-libgsm --enable-libmp3lame --enable-librtmp --enable-libopencv --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libschroedinger --enable-libspeex --enable-libtheora --enable-vaapi --enable-vdpau --enable-libvorbis --enable-libvpx --enable-zlib --enable-gpl --enable-postproc --enable-libcdio --enable-x11grab --enable-libx264 --shlibdir=/usr/lib/x86_64-linux-gnu --enable-shared --disable-static
  libavutil      52. 18.100 / 52. 18.100
  libavcodec     54. 92.100 / 54. 92.100
  libavformat    54. 63.104 / 54. 63.104
  libavdevice    53.  5.103 / 53.  5.103
  libavfilter     3. 42.103 /  3. 42.103
  libswscale      2.  2.100 /  2.  2.100
  libswresample   0. 17.102 /  0. 17.102
  libpostproc    52.  2.100 / 52.  2.100
[video4linux2,v4l2 @ 0x1d682c0] Not a video capture device.
/dev/video0: No such device

I need to inject a video for a local streaming that i will see as buildin video camera using chrome.

It works really fine with firefox not google-chrome yet.

6. Check with google-chrome :

Url for the test : https://opentokrtc.com/

The result is that i don't see the cam.

Thanks a lot for any other idea

thierryg, 2015-01-17:
Hi, thanks for your previous reply. I have seen the bug #40
<https://github.com/umlaeute/v4l2loopback/issues/40> but it was not working
for me.

Here is all the process i have followed for my test ; tell me is something
is missing :

   1. Build :

make clean && make

   1. Check parameters :

modinfo v4l2loopback
filename: /lib/modules/3.13.0-37-generic/extra/v4l2loopback.ko
license: GPL
author: Vasily Levin, IOhannes m zmoelnig zmoelnig@iem.at,Stefan
Diewald,Anton Novikov
description: V4L2 loopback video device
srcversion: DD5EF63321657CC1C01E35A
depends: videodev
vermagic: 3.13.0-37-generic SMP mod_unload modversions
parm: debug:debugging level (higher values == more verbose) (int)
parm: max_buffers:how many buffers should be allocated (int)
parm: max_openers:how many users can open loopback device (int)
parm: devices:how many devices should be created (int)
parm: video_nr:video device numbers (-1=auto, 0=/dev/video0, etc.) (array
of int)
parm: card_label:card labels for every device (array of charp)
parm: exclusive_caps:whether to announce OUTPUT/CAPTURE capabilities
exclusively or not (array of bool)
parm: max_width:maximum frame width (int)
parm: max_height:maximum frame height (int)

   1. Unload previous :

rmmod -r v4l2loopback
or
modprove -r v4l2loopback

   1. Load the current drivers :

sudo su
modprobe videodev
insmod ./v4l2loopback.ko devices=1 video_nr=2 exclusive_caps=1

   1. Check configuration

tree /sys/module/v4l2loopback/
/sys/module/v4l2loopback/
├── coresize
├── holders
├── initsize
├── initstate
├── notes
├── parameters
│   ├── debug
│   ├── exclusive_caps
│   ├── max_buffers
│   ├── max_height
│   ├── max_openers
│   ├── max_width
│   └── video_nr
├── refcnt
├── sections
│   ├── __bug_table
│   ├── __mcount_loc
│   └── __param
├── srcversion
├── taint
└── uevent

4 directories, 17 files
cat /sys/module/v4l2loopback/parameters/exclusive_caps

Y,Y,Y,Y,Y,Y,Y,Y
ls -al /dev/video*

crw-rw----+ 1 root video 81, 0 janv. 17 15:59 /dev/video0
ffmpeg -f v4l2 -list_formats all -i /dev/video0

ffmpeg version 1.2.6-7:1.2.6-1~trusty1 Copyright (c) 2000-2014 the FFmpeg
developers
built on Apr 26 2014 18:52:58 with gcc 4.8 (Ubuntu 4.8.2-19ubuntu1)
configuration: --arch=amd64 --disable-stripping --enable-avresample
--enable-pthreads --enable-runtime-cpudetect
--extra-version='7:1.2.6-1~trusty1' --libdir=/usr/lib/x86_64-linux-gnu
--prefix=/usr --enable-bzlib --enable-libdc1394 --enable-libfreetype
--enable-frei0r --enable-gnutls --enable-libgsm --enable-libmp3lame
--enable-librtmp --enable-libopencv --enable-libopenjpeg --enable-libopus
--enable-libpulse --enable-libschroedinger --enable-libspeex
--enable-libtheora --enable-vaapi --enable-vdpau --enable-libvorbis
--enable-libvpx --enable-zlib --enable-gpl --enable-postproc
--enable-libcdio --enable-x11grab --enable-libx264
--shlibdir=/usr/lib/x86_64-linux-gnu --enable-shared --disable-static
libavutil 52. 18.100 / 52. 18.100
libavcodec 54. 92.100 / 54. 92.100
libavformat 54. 63.104 / 54. 63.104
libavdevice 53. 5.103 / 53. 5.103
libavfilter 3. 42.103 / 3. 42.103
libswscale 2. 2.100 / 2. 2.100
libswresample 0. 17.102 / 0. 17.102
libpostproc 52. 2.100 / 52. 2.100
[video4linux2,v4l2 @ 0x1d682c0] Not a video capture device.
/dev/video0: No such device

I need to inject a video for a local streaming that i will see as buildin
video camera using chrome. for that stuff i am using ffmpeg :

ffmpeg -f mjpeg -i http://<ip>:<port> -f v4l2 /dev/video2

It works really fine with firefox not google-chrome yet.

   1. Check with the latest release of google-chrome :

Url for the test : https://opentokrtc.com/

The result is that i don't see the cam.

Thanks a lot for any other idea

Regards

Thierry GAYET
Mobile : +33.663.849.589
Skype : terranova44 / tux35220
Member of the GNU/Linux Fundation :
https://www.linuxfoundation.org/users/thierryg35
www.nextinnovation.org
www.apprendreindonesien.com

umlaeute, 2015-01-18:
can i remove one of the two last comments (they seem to be identical apart from formatting)?

umlaeute, 2015-01-18:
so the error you get is basically:

    $ ffmpeg -f v4l2 -list_formats all -i /dev/video0
     [...]
     [video4linux2,v4l2 @ 0x1d682c0] Not a video capture device.
     /dev/video0: No such device

which is very much expected: `ffmpeg` tries to *capture* (read) from the device. but if yo uare using the `exclusive_caps` option, then you can **only** capture if you have started a *producer* (writer) beforehand.

what you should do is (in the given order=:

- load the `v4l2loopback` module
- start a producer, like

   `$  gst-launch-0.10 -v videotestsrc ! "video/x-raw-yuv,width=640,height=360,framerate=30/1,format=(fourcc)I420" ! v4l2sink device=/dev/video0`

- start a consumer, like

    `$ ffmpeg -f v4l2 -list_formats all -i /dev/video0`

closed 2015-01-18 by umlaeute

camwhite, 2018-10-08:
This has appeared for me after upgrading to 18.04 LTS. Anyone have any ideas on a fix?

chdsbd, 2019-02-28:
@camwhite Did you resolve this issue? I think I may have just run into this, but it could just be a misconfiguration on my part. I can open the virtual devices in VLC, but Chrome doesn't provide any options.

adrianimajion, 2019-05-14:
I am also encountering this problem, and following @umleaute 's directions did not resolve it. The virtual device is visible in other applications like mplayer, but not in Chrome stable or Chromium. 

@camwhite and @chdsbd , were either of you able to resolve? If not can we re-open this issue?

umlaeute, 2019-05-15:
@adrianimajion why would you want to re-open a bug in `v4l2loopback` if there's a bug  in chrome?

adrianimajion, 2019-05-15:
@umlaeute You're exactly right, my misunderstanding! 

For the other folks in this thread and anyone else who arrives here researching this problem, here's the bug in the Chromium issue tracker: https://bugs.chromium.org/p/chromium/issues/detail?id=757399

josx, 2020-04-17:
Here my case:

```
insmod v4l2loopback video_nr=4 exclusive_caps=1
ffmpeg -i tcp://127.0.0.1:2000?listen -f v4l2 /dev/video4
```
Then I am ingesting from OBS.

Works with well on Firefox, and in other any app like ```ffplay /dev/video4```or `` cheese```

Not on Chromium Versión 80.0.3987.116 or Chrome Versión 83.0.4100.3

umlaeute, 2020-04-17:
@josx is your prolem a duplicate of #274 (check the advise there, and report back (on ticket #274) whether this solved your problem))

theapache64, 2020-04-19:
Same issue for me. I am using `Ubuntu 16.04`. It works fine with `firefox`, but chrome can't detect the device. 

Note:  I use ffmpeg to feed the webcam. `ffmpeg -i video.mp4 -f v4l2 /dev/video0` 

Any help ?