v4l2loopback/v4l2loopback #79: Ops on 3.19

diraimondo, 2015-02-10:
On a fresh built 3.19 kernel it gives an ops on the kernel log, but it looks to work:
```
[    2.550122] v4l2loopback driver version 0.8.0 loaded                                                                                                                 [0/1995]
[    2.551026] ------------[ cut here ]------------
[    2.551033] WARNING: CPU: 2 PID: 1174 at drivers/media/v4l2-core/v4l2-ioctl.c:1025 v4l_querycap+0x3b/0x70 [videodev]()
[    2.551034] Modules linked in: v4l2loopback(O) videodev media w83627ehf hwmon_vid loop dm_crypt iTCO_wdt nuvoton_cir x86_pkg_temp_thermal iTCO_vendor_support snd_hda_codec_hdmi intel_powerclamp evdev rc_core intel_rapl parport_pc iosf_mbi parport battery kvm_intel kvm pcspkr psmouse i2c_i801 tpm_tis serio_raw tpm snd_hda_intel i915 snd_hda_controller snd_hda_codec video snd_hwdep drm_kms_helper snd_pcm drm snd_timer snd xhci_pci i2c_algo_bit mei_me lpc_ich soundcore xhci_hcd mei i2c_core mfd_core button processor thermal_sys btrfs xor raid6_pq dm_mod sg sd_mod ahci libahci libata crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel alx mdio aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd scsi_mod ehci_pci ehci_hcd usbcore usb_common
[    2.551065] CPU: 2 PID: 1174 Comm: v4l_id Tainted: G           O   3.19.0-my #1
[    2.551066] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./B85M, BIOS P2.10 12/06/2013
[    2.551067]  0000000000000000 ffffffffa05ce090 ffffffff8152eeba 0000000000000000
[    2.551069]  ffffffff8106ae90 ffff880211217dc0 0000000080685600 0000000000000000
[    2.551070]  ffff8802120f7200 0000000000000000 ffffffffa05b978b 0000000000000000
[    2.551071] Call Trace:
[    2.551076]  [<ffffffff8152eeba>] ? dump_stack+0x40/0x50
[    2.551079]  [<ffffffff8106ae90>] ? warn_slowpath_common+0x90/0xe0
[    2.551082]  [<ffffffffa05b978b>] ? v4l_querycap+0x3b/0x70 [videodev]
[    2.551084]  [<ffffffffa05b9075>] ? __video_do_ioctl+0x295/0x310 [videodev]
[    2.551086]  [<ffffffff811b5ee9>] ? __inode_permission+0x29/0xc0
[    2.551088]  [<ffffffff811b70e1>] ? link_path_walk+0x61/0x850
[    2.551090]  [<ffffffffa05b99d1>] ? video_usercopy+0x211/0x540 [videodev]
[    2.551091]  [<ffffffffa05b8de0>] ? v4l_printk_ioctl+0xa0/0xa0 [videodev]
[    2.551093]  [<ffffffffa05b5673>] ? v4l2_ioctl+0x123/0x160 [videodev]
[    2.551095]  [<ffffffff811be12e>] ? do_vfs_ioctl+0x7e/0x540
[    2.551097]  [<ffffffff811be689>] ? SyS_ioctl+0x99/0xb0
[    2.551099]  [<ffffffff8153496d>] ? system_call_fastpath+0x16/0x1b
[    2.551100] ---[ end trace 0beebf19de5a25c0 ]---
[    2.552456] ------------[ cut here ]------------
```

PhobosK, 2015-03-07:
I can confirm that too...
Same call trace on Gentoo **kernel 3.19** and **v4l2loopback master** as of today
```
v4l2loopback driver version 0.8.0 loaded
------------[ cut here ]------------
WARNING: CPU: 1 PID: 3081 at drivers/media/v4l2-core/v4l2-ioctl.c:1025 v4l_querycap+0x41/0x52 [videodev]()
Modules linked in: v4l2loopback(O) ip6table_filter ip6_tables cls_basic cls_flow cls_fw cls_u32 sch_tbf sch_prio sch_htb sch_hfsc sch_ingress sch_sfq xt_CHECKSUM ipt_rpfilter xt_statistic xt_CT xt_connlimit xt_realm xt_addrtype ip_set_hash_ip xt_comment xt_recent xt_nat ipt_REJECT nf_reject_ipv4 ipt_MASQUERADE nf_nat_masquerade_ipv4 ipt_ECN ipt_CLUSTERIP ipt_ah xt_set ip_set xt_NFLOG nfnetlink_log xt_LOG nf_log_ipv4 nf_log_common nf_nat_tftp nf_nat_snmp_basic nf_conntrack_snmp nf_nat_sip nf_nat_pptp nf_nat_proto_gre nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda ts_kmp nf_conntrack_amanda nf_conntrack_sane nf_conntrack_tftp nf_conntrack_sip nf_conntrack_proto_udplite nf_conntrack_proto_sctp nf_conntrack_pptp nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_broadcast nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp xt_TPROXY nf_defrag_ipv6 xt_time xt_TCPMSS xt_tcpmss xt_sctp xt_policy xt_pkttype xt_physdev br_netfilter bridge stp llc xt_owner xt_NFQUEUE xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_helper xt_hashlimit xt_DSCP xt_dscp xt_dccp xt_conntrack xt_connmark xt_CLASSIFY xt_tcpudp xt_state iptable_raw iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_conntrack iptable_mangle nfnetlink iptable_filter ip_tables x_tables tun it87 hwmon_vid snd_usb_audio snd_usbmidi_lib snd_rawmidi hid_logitech_hidpp hid_logitech_dj snd_seq_device snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic pcspkr nvidia(PO) serio_raw k10temp snd_hda_intel ata_generic pata_acpi snd_hda_controller snd_hda_codec wmi drm floppy button sp5100_tco snd_hwdep snd_pcm snd_timer snd xhci_pci algif_skcipher af_alg sch_fq_codel vboxnetadp(O) vboxnetflt(O) vboxdrv(O) binfmt_misc cpufreq_userspace cpufreq_powersave cpufreq_ondemand cpufreq_conservative loop hangcheck_timer uinput btusb bluetooth rfkill dm_zero dm_thin_pool dm_persistent_data dm_bio_prison dm_service_time dm_round_robin dm_multipath virtio_pci virtio_scsi virtio_blk virtio_net virtio_console virtio_balloon virtio_ring virtio iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi fuse xfs libcrc32c nfs lockd grace sunrpc fscache jfs reiserfs btrfs ext2 multipath linear raid0 dm_raid raid456 async_raid6_recov async_memcpy raid1 raid10 async_pq async_xor xor async_tx raid6_pq dm_snapshot dm_bufio dm_crypt dm_mirror dm_region_hash dm_log dm_mod hid_sony hid_microsoft hid_logitech ff_memless hid_a4tech uhci_hcd usb_storage scsi_transport_fc sg pata_amd pata_platform usbhid uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common videodev media i2c_core ohci_pci kvm_amd kvm firewire_ohci pata_jmicron firewire_core ahci libahci xhci_hcd ehci_pci ehci_hcd ohci_hcd r8169 mii libata usbcore usb_common acpi_cpufreq processor
CPU: 1 PID: 3081 Comm: v4l_id Tainted: P           O   3.19.0-gentoo #1
Hardware name: Gigabyte Technology Co., Ltd. GA-890GPA-UD3H/GA-890GPA-UD3H, BIOS FF 11/24/2010
0000000000000000 0000000000000009 ffffffff8143ca2a 0000000000000000
ffffffff8107770a 0000000000000001 ffffffffa041ddf8 ffff8803e1857df8
ffff8803e1857df8 ffffffffa042e9b0 0000000000000000 ffff8803e1857d00
Call Trace:
[<ffffffff8143ca2a>] ? dump_stack+0x47/0x67
[<ffffffff8107770a>] ? warn_slowpath_common+0x93/0xab
[<ffffffffa041ddf8>] ? v4l_querycap+0x41/0x52 [videodev]
[<ffffffffa041ddf8>] ? v4l_querycap+0x41/0x52 [videodev]
[<ffffffffa041f619>] ? __video_do_ioctl+0x177/0x258 [videodev]
[<ffffffffa041f3b2>] ? video_usercopy+0x22a/0x30e [videodev]
[<ffffffffa041f4a2>] ? video_ioctl2+0xc/0xc [videodev]
[<ffffffffa041b4bd>] ? v4l2_ioctl+0x74/0xec [videodev]
[<ffffffff811411c1>] ? do_vfs_ioctl+0x367/0x42b
[<ffffffff81148bd6>] ? __fd_install+0x1c/0x4e
[<ffffffff81121d60>] ? kmem_cache_free+0xa3/0x10f
[<ffffffff81133773>] ? do_sys_open+0xd7/0xe9
[<ffffffff811412b8>] ? SyS_ioctl+0x33/0x58
[<ffffffff81440a16>] ? system_call_fastpath+0x16/0x1b
---[ end trace ad5a21623839960a ]---
```

yukkeorg, 2015-03-09:
Probably this issue related about this.
https://www.mail-archive.com/linux-media@vger.kernel.org/msg82848.html

umlaeute, 2015-03-19:
assuming that yukkeorg's patch really fixed this.

closed 2015-03-19 by umlaeute