v4l2loopback/v4l2loopback #83: videotestsrc not working with gstreamer 1.0

atoy40, 2015-03-19:
Hello,

I'm unable to launch a simple gstreamer pipeline with the testy source
Here is the result (i've just modprobe the module and then launch the following command):

```
gst-launch-1.0  videotestsrc ! v4l2sink device=/dev/video1 
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2933): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming task paused, reason error (-5)
Execution ended after 0:00:00.066918471
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
 with a GST_DEBUG=3, i found this error :
```
0:00:00.083227552 26991      0x23e4e30 ERROR                   v4l2 gstv4l2bufferpool.c:1132:gst_v4l2_buffer_pool_dqbuf:<v4l2sink0:pool:sink> No free buffer found in the pool at index 1.
0:00:00.083285313 26991      0x23e4e30 WARN                 basesrc gstbasesrc.c:2933:gst_base_src_loop:<videotestsrc0> error: Internal data flow error.
0:00:00.083295272 26991      0x23e4e30 WARN                 basesrc gstbasesrc.c:2933:gst_base_src_loop:<videotestsrc0> error: streaming task paused, reason error (-5)
```

Im' using the version 0.8.0 packaged in ubuntu 14.10 with a 3.16.0 kernel

thank you very much.

Anthony.

yukkeorg, 2015-03-20:
i think this problem is related to #60.

bsenicar, 2015-06-26:
Any news on that issue?

I like to use gstreamer and using 0.10 branch is not really an option since it is long time appreciated.

BR Borut

Adwind, 2015-07-20:
To make gstreamer 1.0 we found a way around .
     1) You start a "client" with gstreamer 1.0 that has the videotestsrc (or other and their decoder) and sends over a udpsink. 

        gst-launch-1.0 -v videotestsrc ! queue ! rtpvrawpay ! udpsink port=51111

abickerton, 2015-08-02:
This is also a problem in debian/sid version 0.9.1-2. 

'''
$gst-launch-0.10 videotestsrc ! v4l2sink device=/dev/video0

ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Failed to query attributes of input 0 in device /dev/video0

Additional debug info:
v4l2_calls.c(134): gst_v4l2_fill_lists (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Failed to get 0 in input enumeration for /dev/video0. (25 - Inappropriate ioctl for device)
Setting pipeline to NULL ...
'''

astronouth7303, 2015-10-30:
I'm getting this on Debian Testing:

* Linux 4.2.0-1-amd64
* v4l2loopback-dkms 0.9.1-3
* gstreamer1.0-plugins-good 1.6.0-1

My error is slightly different:
```
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.034374233 16672      0x1a40d40 ERROR          v4l2allocator gstv4l2allocator.c:1299:gst_v4l2_allocator_dqbuf:<v4l2sink0:pool:sink:allocator> buffer 2 was not queued, this indicate a driver bug.
0:00:00.034428360 16672      0x1a40d40 WARN                 basesrc gstbasesrc.c:2943:gst_base_src_loop:<videotestsrc0> error: Internal data flow error.
0:00:00.034446519 16672      0x1a40d40 WARN                 basesrc gstbasesrc.c:2943:gst_base_src_loop:<videotestsrc0> error: streaming task paused, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data flow error.
```

atamariya, 2015-11-10:
Try with my patch https://github.com/umlaeute/v4l2loopback/pull/100 .

bsenicar, 2015-11-10:
Still getting error as shown in astronouth7303 post.

BR Borut

JulienMichot, 2015-11-23:
Same on latest Ubuntu :(

dankasak, 2015-11-25:
Me too :( Gentoo.

umlaeute, 2015-11-25:
patches welcome

ndufresne, 2016-10-20:
To all, v4l2loopback driver has important bugs in it's queue management. It's probably reporting some random buffer index. Since 1.0 we do validate this stuff since otherwise it can crash GStreamer. We also use larger queues to avoid doing buffer copied (while in 0.10 we always copy). For anything where the v4l2 queue is bigger then 2 frames, the driver output is wrong. A simple workaround is to use tee multiplexer, which forces a copy:

  gst-launch-1.0 videotestsrc ! tee ! v4l2sink device=...

Long term, someone need to fix the driver.

umlaeute, 2016-10-20:
given my current time allocation, i'm happily accepting patches.

silencefall, 2016-12-01:
Hi
I also met this issue. I seems fixed with this patch
https://github.com/silencefall/v4l2loopback/commit/97dc86bcddd74cc97254bbca0cebbd55569c3cc6
Test environment:
ubuntu 14.04, gstreamer 10.1
Test pipeline:
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video1
gst-launch-1.0 v4l2src device=/dev/video1 ! glimagesink

umlaeute, 2016-12-02:
hopefully fixed in `0.10.0`

closed 2016-12-02 by umlaeute

ndufresne, 2017-01-16:
I just rested, the fix is kind of partial. The suggested pipeline now works, but now we fail on the receiver side:
  gst-launch-1.0 videotestsrc ! tee ! v4l2sink
  gst-launch-1.0 v4l2src ! videoconvert ! autovideosink
     gstv4l2allocator.c:1299:gst_v4l2_allocator_dqbuf:<v4l2src0:pool:src:allocator> buffer 1 was not queued, this indicate a driver bug.

baptx, 2017-11-09:
I have to use the "tee" workaround on Xubuntu 16.04 LTS, you should add the example `gst-launch-1.0 -v videotestsrc ! tee ! v4l2sink device=/dev/video1` in the wiki if other people get the "Internal data flow error".

ndufresne, 2017-11-09:
This will no longer work in 1.14, in replacement, you'll be able to workaround this driver bug with "identity drop-allocation=1".

baptx, 2017-11-14:
Good to know. Is there a way to fix the driver bug directly so we don't have to use a workaround?

ndufresne, 2017-11-14:
Sure, go ahead ;-P

darenw, 2018-03-08:
I'm trying to get a simple pipeline to work, sending video to v4l2sink.  I can use videotestsrc just fine.  It needed the 'tee' to work.   But a specialized camera source isn't working at all.  With or without tee.  With the identity drop-allocation=1, I get:

WARNING: erroneous pipeline: no property "drop-allocation" in element "identity0"

???

umlaeute, 2018-03-09:
@darenw please make sure that your pipeline is actually valid before posting problems (if GSTreamer's `identity` element doesn't have a `drop-allocation` property, then you should read the documentation on `identity` first)

thirdeye-peter, 2018-06-06:
@darenw: `drop-allocation` is only available in gstreamer 1.14.x (and 1.13.x), it was only added recently

https://github.com/GStreamer/gstreamer/commit/2cc5c539a3afb8d887990b22fc62a43111281cb5

homespring, 2018-08-14:
I don't know why this issue is closed... I'm using Xubuntu 18.04 with Gstreamer 1.14.1 and when I run pipeline:

 gst-launch-1.0 videotestsrc ! identity drop-allocation=1 ! v4l2sink device=/dev/video100

I can't see any test video on /dev/video100. This makes v4l2loopback unusable on newer systems as I'm forced to use Xubuntu 16.04 with Gstreamer 1.8.3 so it can still work.

raymanfx, 2019-11-04:
Please reopen.
I'm on Fedora 31 with GST version: 1.16.1.
Running the following minimal example yields the "streaming stopped, reason not-negotiated (-4)" error. There's no way to work around it apparently, since 'tee' no longer works and GST 0.10 is broken in Fedora (packages deps cannot be resolved).

source:
`gst-launch-1.0 videotestsrc ! identity drop-allocation=1 ! v4l2sink device=/dev/video9`

sink:
`gst-launch-1.0 v4l2src device=/dev/video9 ! autovideosink`

umlaeute, 2019-11-04:
@raymanfx it works here on Derbian with gstreamer-1.16.1

- which of the two commands throws the error?
- which version of `v4l2loopback` do you use?

raymanfx, 2019-11-04:
Thanks for the quick answer.
It's always the sink pipeline that throws the error.
I'm on HEAD (94dcdf3de017ef39b005ca1864772dbf93a143a3).

umlaeute, 2019-11-04:
try loading the module with `buffesr=2` (or `buffers=1`):

    sudo rmmod v4l2loopback
    sudo modprobe v4l2loopback buffers=2

(and use `identity drop-allocation=1` before `v4l2sink` in the consumer)

raymanfx, 2019-11-04:
Neither combination is working, it's always the same error.
I tried buffers=1, buffers=2, and the drop-allocation attribute (basically all the samples listed in my previous comment).

umlaeute, 2019-11-04:
you need to both set `buffer=2` and use `drop-allocation` (you probably did that, i just want to make sure)

raymanfx, 2019-11-04:
Yeah, I did that, but it didn't help.

marcoippolito, 2020-03-16:
I'm having, I think, the same problem with Ubuntu 18.04.4 Desktop:

    (base) marco@pc01:~$ cheese

    (cheese:18986): Gtk-WARNING **: 14:12:02.085: Theme parsing error: cheese.css:7:35: The style
    property GtkScrollbar:min-slider-length is deprecated and shouldn't be used anymore. It will be 
    removed in a future version

    (cheese:18986): cheese-WARNING **: 14:12:03.321: Failed to allocate a buffer: gstv4l2src.c(998):
    gst_v4l2src_create (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source
    /GstBin:bin18/GstV4l2Src:v4l2src1

colin-adams, 2020-03-20:
I'm having the same problem on Fedora - I have gstreamer 0.10 installed:

gstreamer-0.10.36-24.fc31.x86_64

I'm trying to use my DSLR as a webcam. Is there any known workaround currently?

colin-adams, 2020-03-20:
@raymanfx Did you ever have any success with Fedora?

raymanfx, 2020-03-20:
No, I could never make it work with recent versions of gstreamer.

umlaeute, 2020-03-20:
@colin-adams this ticket is explicitely about `GStreamer-1.0`, not about `GStreamer-0.10`. you can open a separate issue if you want to.

drmichaeljgruber, 2020-03-30:
Fedora 31, gstreamer 1.16.2
`gst-launch-1.0  videotestsrc ! v4l2sink device=/dev/video1` works whether I load the module without parameters or limitting the buffers.
[I have a different problem, though, which does not belong in this issue.]

umlaeute, 2020-03-30:
@drmichaeljgruber could you also specify which version of `v4l2loopback` you are using (just update your comment)

zingfrid, 2020-04-01:
Just leave it here, because I've spent half a day making it works!
`sudo modprobe v4l2loopback buffers=2`

`gst-launch-1.0 rtspsrc location=rtsp://admin:admin@192.168.1.12 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! "video/x-raw,format=YUY2" ! tee ! v4l2sink device=/dev/video0`

umlaeute, 2020-04-01:
that should be `max_buffers=2` (and this is the default since e4f2c3c2707bd7b7ce0cfe9bd582bda4e94f58ff

stephematician, 2025-01-31:
@ndufresne I'm also stuck on the following:

```sh
# can use a sink when max_buffers=4 no problem
sudo modprobe v4l2loopback max_buffers=4 video_nr=10
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video10
# open a new terminal and try to use the v4l2-loopback node as a source
gst-launch-1.0 v4l2src device=/dev/video10 ! autovideosink
```

Results in:

```console
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3177): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.000230347
Setting pipeline to NULL ...
Freeing pipeline ...
```

Attached the log with `GST_DEBUG=4`; it gets as far as enumerating frame intervals - then it fails. Using `debug=2` with v4l2-loopback only showed that the final IOCTL was ENUM_FRAMEINTERVAL. It's like gstreamer gets all the information it wants about the loopback device - then decides nope! (Wondering if its something to do with trying interlaced mode?)

[gst_v4l2loopback_experiment.txt](https://github.com/user-attachments/files/18613209/gst_v4l2loopback_experiment.txt)

stephematician, 2025-02-01:
> # can use a sink when max_buffers=4 no problem
> sudo modprobe v4l2loopback max_buffers=4 video_nr=10
> gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video10
> # open a new terminal and try to use the v4l2-loopback node as a source
> gst-launch-1.0 v4l2src device=/dev/video10 ! autovideosink

I made a mistake - with `videoconvert` pad this works.