v4l2loopback/v4l2loopback #84: Raspberry pi fail to install

alduxvm, 2015-03-19:
Hi!

when doing the "sudo make" and "sudo make install" after installing linux headers for 3.2.0-4-rpi, and linking it to the build directory, now my error is:

warning: you may need to install module-init-tools

and of course I have module init tools... and it does not install it. Did I install the incorrect linux headers?

cheers!

umlaeute, 2015-03-20:
a few things:
- you must of course install the linux-headers that match your current kernel-version (e.g. as reported by `uname -r`). I personally have no clue which kernel your rpi-distro is running (nor which rpi-distro you are using)
- what do you mean by "linking it [the headers] to the build directory"? 
- in any case, `linux-3.2.0` is really a rather old kernel, are you sure that this is correct for your system?
- don't use `sudo` for building the driver; a simple `make` shall be enough (though you need `sudo` to run `make install`)
- after running `make`, do you have a `v4l2loopback.ko` file?
- the "error" you get is really a *warning* (as it says)
- the warning is issued by "something else" (not by v4l2loopback's build system); try to find out what is issuing it (a wild guess: `make install` calls `depmod`; iirc in the olde days `depmod` was provided by `module-init-tools` (these days its part of `kmod`))
- what are your expectations with "installing it"?

alduxvm, 2015-03-21:
I have 3.18.7+ but my linux-headers are empty, so I installed those headers via apt-get in order to have the "build" folder, and then link it and then do the make.

After doing it, I indeed have the v4l2loopback.ko file...

What I want to achieve is having a /dev/videoXX driver and then be able to do this: https://github.com/PureEngineering/LeptonModule/tree/master/flirpi

umlaeute, 2015-03-22:
well, you have to load the module.

check the *RUN* section in the  [README](/umlaeute/v4l2loopback).

closed 2015-03-22 by umlaeute

gcs278, 2015-04-13:
Hey @alduxvm, did you solve this issue

I am having the same exact problems as you, using 3.18.7+ that doesn't contain build directory so I link it to 3.2.0-4-rpi's build directory. I get v4l2loopback.ko when I make, but make install doesn't properly install it; therefore modprobe can't find v4l2loopback.

Any suggestions?

umlaeute, 2015-04-13:
@gcs278  you *must not* build a kernel-module using *headers-X* if you want to use it with *kernel-Y* (if X != Y). put otherwise: a module built for *kernel-X* is *not* be compatible with *kernel-Y*.

what does **not properly installed** mean? a module build for *kernel-X* should/will be installed at the module-location for *kernel-X* (but obviously not at the module-location for *kernel-Y*).

you don't even need to install the kernel-module to use it (with the correct kernel).
just run `insmod ./v4l2loopback.ko` or `make modprobe` (which does the same).

in any case: please install the headers matching your kernel.

gcs278, 2015-04-13:
@umlaeute Thanks for the quick response.
Okay yea that would make sense. I need to build it with my kernel's version.

I'm running 3.18.7+ and I assume `sudo apt-get install linux-headers-3.18.0-trunk-rpi2` are the headers I need. I've installed them and then did a `sudo ln -sfn /lib/modules/3.18.0-trunk-rpi2/build /lib/modules/3.18.7-v7+/build` to link the build directory.

But when I run `make`, I get this:
<pre><code>make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback-0.8.0 modules
make[1]: Entering directory '/usr/src/linux-headers-3.18.0-trunk-rpi2'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
  CC [M]  /home/pi/v4l2loopback-0.8.0/v4l2loopback.o
/home/pi/v4l2loopback-0.8.0/v4l2loopback.c: In function ‘attr_store_maxopeners’:
/home/pi/v4l2loopback-0.8.0/v4l2loopback.c:486:2: error: implicit declaration of function ‘strict_strtoul’ [-Werror=implicit-function-declaration]
cc1: some warnings being treated as errors
/usr/src/linux-headers-3.18.0-trunk-common/scripts/Makefile.build:268: recipe for target '/home/pi/v4l2loopback-0.8.0/v4l2loopback.o' failed
make[4]: *** [/home/pi/v4l2loopback-0.8.0/v4l2loopback.o] Error 1
/usr/src/linux-headers-3.18.0-trunk-common/Makefile:1398: recipe for target '_module_/home/pi/v4l2loopback-0.8.0' failed
make[3]: *** [_module_/home/pi/v4l2loopback-0.8.0] Error 2
Makefile:145: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.18.0-trunk-rpi2'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2</code></pre>

I'm pretty confused by this. It does however build correctly with linux-headers-3.2.0-4-rpi.

umlaeute, 2015-04-13:
hmm, no `3.18.0-trunk-rpi` is *not* the same kernel-version as `3.18.7-v7+`.
if you install the correct headers, then you should not need to do any weird symlinks.

please consult the `raspbian` maintainers how to get the proper headers for your kernel.

gcs278, 2015-04-13:
Alright so I figure everything out, here's a summary how I got it to work on for the Raspberry Pi2 for anyone that hits this problem:

Rasbian doesn't ship with the Linux-Headers in 3.18.7-v7+ (aka the /build folder). You have two choices: build the kernel from source (http://github.com/raspberrypi/linux) to get these headers or just revert to 3.18.0.

It was a lot easier to revert to 3.18.0. Run these commands to switch to 3.18.0:
<pre><code>
sudo aptitude install -y linux-image-rpi2-rpfv linux-headers-rpi2-rpfv
echo -e "kernel=vmlinuz-3.18.0-trunk-rpi2\ninitramfs initrd.img-3.18.0-trunk-rpi followkernel" | sudo tee -a /boot/config.txt
sudo reboot
</pre></code>

Rasbian will now boot up using linux kernel 3.18.0.

Now you can run <code>sudo make</code> and then <code>sudo insmod ./v4l2loopback.ko</code> and bam you should see another /dev/video pop up. I tried to use this guy's packages for linux-headers (http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/), but the 3.18.7-v7+ package actually compiled for 3.18.7-v7 for some reason so I didn't work for me.

Thanks @umlaeute

druadan, 2015-04-21:
You forgot equals sign next to initramfs in echo command.

rsegecin, 2016-05-20:
Sorry if I should've open in another thread. 

Hi gcs278 I'm trying to accomplish the same thing to build modprobe and run it on my raspberry pi, I followed your instructions:

`sudo aptitude install -y linux-image-rpi2-rpfv linux-headers-rpi2-rpfv
echo -e "kernel=vmlinuz-3.18.0-trunk-rpi2\ninitramfs=initrd.img-3.18.0-trunk-rpi followkernel" | sudo tee -a /boot/config.txt
sudo reboot`

but after I rebooted all the files in /boot/* were gone and apparently only the files in this directory were gone.  My raspberry is "Pi 2 Model B". Is there any way to restore these files?