v4l2loopback/v4l2loopback #85: changes for kernel 3.16 replace strict_strto* call with kstrto*

moosbaue, 2015-03-22:
Hello

line 506 v4l2loopback.c reads
if (strict_strtoul(buf, 0, &curr))

This must be replaced by
if (kstrtoul(buf, 0, &curr))
from kernel 3.16 onward

kind regards
Josef

umlaeute, 2015-03-22:
thanks.

afaict, this should read kernel *3.18*

it has been fixed a few months ago with 21195cd6.

closed 2015-03-22 by umlaeute

moosbaue, 2015-03-22:
sorry, did not read carefully enough

umlaeute, 2015-03-24:
np. in any case thanks for the suggested fix (even though it was not needed this time...)

moosbaue, 2015-03-25:
and the version numbering confused me; had 0.8.0 and is still 0.8.0
Imho maybe as a suggestion even though bug fixing, increasing versions are helpfull to some people

umlaeute, 2015-03-25:
versions are increased for each release.
without release, versions are not increased.