v4l2loopback/v4l2loopback #87: Failed to build on kernel 3.19.2-1-ARCH

notnola, 2015-04-06:
[nola@TimeMachine v4l2loopback-master]$ make install
make -C /lib/modules/`uname -r`/build M=/home/nola/Downloads/v4l2loopback-master modules_install
make[1]: Entering directory '/usr/lib/modules/3.19.2-1-ARCH/build'
make[1]: *** No rule to make target 'modules_install'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/3.19.2-1-ARCH/build'
Makefile:47: recipe for target 'install' failed
make: *** [install] Error 2

umlaeute, 2015-04-06:
can this be related to a not-properly configured kernel-header/source tree?

you *must* have the kernel-headers for the target-kernel (usually the kernel you are running) installed on configured on your build system.

notnola, 2015-04-06:
I am pretty sure I had installed the proper headers. Returning a different error now. Keep in mind, I am not to skilled in this area, but am trying my best.

[nola@TimeMachine v4l2loopback-master]$ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/nola/Downloads/v4l2loopback-master modules
make[1]: Entering directory '/usr/lib/modules/3.19.2-1-ARCH/build'
  CC [M]  /home/nola/Downloads/v4l2loopback-master/v4l2loopback.o
/home/nola/Downloads/v4l2loopback-master/v4l2loopback.c: In function ‘v4l2loopback_s_ctrl’:
/home/nola/Downloads/v4l2loopback-master/v4l2loopback.c:1224:50: error: ‘struct v4l2_ctrl’ has no member named ‘val64’
  return v4l2loopback_set_ctrl(dev, ctrl->id, ctrl->val64);
                                                  ^
/home/nola/Downloads/v4l2loopback-master/v4l2loopback.c:1225:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
scripts/Makefile.build:263: recipe for target '/home/nola/Downloads/v4l2loopback-master/v4l2loopback.o' failed
make[2]: *** [/home/nola/Downloads/v4l2loopback-master/v4l2loopback.o] Error 1
Makefile:1382: recipe for target '_module_/home/nola/Downloads/v4l2loopback-master' failed
make[1]: *** [_module_/home/nola/Downloads/v4l2loopback-master] Error 2
make[1]: Leaving directory '/usr/lib/modules/3.19.2-1-ARCH/build'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2

umlaeute, 2015-04-07:
two errors are unrelated, please open (in the future at least) a separate issue.

anyhow: the errors indicates that you are not using the latest v4l2loopback but some older version which one? your directory indicates that you *downloaded* the master branch (rather than doing a `git clone/pull`) at some time in the past.
please try the latest [master](/umlaeute/Gem/tree/master)

nolanryan22, 2015-04-09:
I am having the same issues on Manjaro, I have the correct linux headers for my kernel: 

Kernel Information:
$ uname -r
3.19.3-1-MANJARO
$ sudo pacman -Qi linux319-headers
[sudo] password for marduk: 
Name           : linux319-headers
Version        : 3.19.3-1
Description    : Header files and scripts for building modules for Linux319 kernel
Architecture   : x86_64
URL            : http://www.kernel.org/
Licenses       : GPL2
Groups         : None
Provides       : linux-headers=3.19.3
Depends On     : None
Optional Deps  : None
Required By    : None
Optional For   : None
Conflicts With : None
Replaces       : None
Installed Size :  43.90 MiB
Packager       : Philip Müller <philm@manjaro.org>
Build Date     : Thu 26 Mar 2015 01:21:15 PM CST
Install Date   : Wed 08 Apr 2015 10:51:47 PM CST
Install Reason : Explicitly installed
Install Script : No
Validated By   : Signature
----------------------------------------------------------------------------------

Error:

$ make                                                                                                                                                                    
Building v4l2-loopback driver...                                                                                                                                                                      
make -C /lib/modules/`uname -r`/build M=/home/marduk/Downloads/v4l2loopback/v4l2loopback modules                                                                                                      
make[1]: Entering directory '/usr/lib/modules/3.19.3-1-MANJARO/build'                                                                                                                                 
  CC [M]  /home/marduk/Downloads/v4l2loopback/v4l2loopback/v4l2loopback.o                                                                                                                             
/home/marduk/Downloads/v4l2loopback/v4l2loopback/v4l2loopback.c: In function ‘v4l2loopback_s_ctrl’:                                                                                                   
/home/marduk/Downloads/v4l2loopback/v4l2loopback/v4l2loopback.c:1224:50: error: ‘struct v4l2_ctrl’ has no member named ‘val64’                                                                        
  return v4l2loopback_set_ctrl(dev, ctrl->id, ctrl->val64);                                                                                                                                           
                                                  ^                                                                                                                                                   
/home/marduk/Downloads/v4l2loopback/v4l2loopback/v4l2loopback.c:1225:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
scripts/Makefile.build:263: recipe for target '/home/marduk/Downloads/v4l2loopback/v4l2loopback/v4l2loopback.o' failed
make[2]: *** [/home/marduk/Downloads/v4l2loopback/v4l2loopback/v4l2loopback.o] Error 1
Makefile:1382: recipe for target '_module_/home/marduk/Downloads/v4l2loopback/v4l2loopback' failed
make[1]: *** [_module_/home/marduk/Downloads/v4l2loopback/v4l2loopback] Error 2
make[1]: Leaving directory '/usr/lib/modules/3.19.3-1-MANJARO/build'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2

closed 2015-04-09 by umlaeute