v4l2loopback/v4l2loopback #93: Inappropriate ioctl

jcaesar, 2015-07-16:
I have trouble doing v4l2loopback-ctl on a loopback device. The core of the problem is

    Setting pipeline to PAUSED ...
    ERROR: Pipeline doesn't want to pause.
    ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Failed to query attributes of input 0 in device /dev/video9
This problem started occurring seemingly at random but has been persisting over reloads and reboots since then.

Here is the output of me trying to -ctl, with some additional info. It also shows that I can write to the device with the script from https://gist.github.com/TimSC/6532334. What it does not show: I can play the video in mplayer/mpv, but not in skype. Also, gst-launch fails with the inappropriate ioctl, too.

    $ v4l2-ctl --all -d /dev/video9
    Driver Info (not using libv4l2):
    	Driver name   : v4l2 loopback
    	Card type     : Loopback Device
    	Bus info      : platform:v4l2loopback-000
    	Driver version: 3.16.7
    	Capabilities  : 0x85008002
    		Video Output
    		Video Memory-to-Memory
    		Read/Write
    		Streaming
    		Device Capabilities
    	Device Caps   : 0x05008002
    		Video Output
    		Video Memory-to-Memory
    		Read/Write
    		Streaming
    Video output: 0 (loopback in)
    Format Video Output:
    	Width/Height  : 0/0
    	Pixel Format  : 'BGR4'
    	Field         : None
    	Bytes per Line: 0
    	Size Image    : 0
    	Colorspace    : SRGB
    Streaming Parameters Video Capture:
    	Frames per second: 30.000 (30/1)
    	Read buffers     : 8
    Streaming Parameters Video Output:
    	Frames per second: 30.000 (30/1)
    	Write buffers    : 8
    
    User Controls
    
                        keep_format (bool)   : default=0 value=0
                  sustain_framerate (bool)   : default=0 value=0
                            timeout (int)    : min=0 max=100000 step=1 default=0 value=0
                   timeout_image_io (bool)   : default=0 value=0
                        keep_format (bool)   : default=0 value=0
                  sustain_framerate (bool)   : default=0 value=0
                            timeout (int)    : min=0 max=100000 step=1 default=0 value=0
                   timeout_image_io (bool)   : default=0 value=0
    $ v4l2loopback-ctl set-caps "video/x-raw-yuv, width=640, height=480" /dev/video9
    Setting pipeline to PAUSED ...
    ERROR: Pipeline doesn't want to pause.
    ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Failed to query attributes of input 0 in device /dev/video9
    Additional debug info:
    v4l2_calls.c(134): gst_v4l2_fill_lists (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
    Failed to get 0 in input enumeration for /dev/video9. (25 - Inappropriate ioctl for device)
    Setting pipeline to NULL ...
    Freeing pipeline ...
    output to /dev/video9 failed
    $ v4l2-ctl --all -d /dev/video9
    [no change]
    toef@fuckvm:~/fd/feedlb$ ./image-to-v4l2loopback.py /dev/video9 
    <open file '/dev/video9', mode 'wr' at 0x7f1b4887e9c0>
    get capabilities result 0
    capabilities 0x85008002L
    v4l2 driver: v4l2 loopback
    set format result 0
    ^C
    KeyboardInterrupt
    toef@fuckvm:~/fd/feedlb$ v4l2-ctl --all -d /dev/video9
    Driver Info (not using libv4l2):
    	Driver name   : v4l2 loopback
    	Card type     : Loopback Device
    	Bus info      : platform:v4l2loopback-000
    	Driver version: 3.16.7
    	Capabilities  : 0x85008003
    		Video Capture
    		Video Output
    		Video Memory-to-Memory
    		Read/Write
    		Streaming
    		Device Capabilities
    	Device Caps   : 0x05008003
    		Video Capture
    		Video Output
    		Video Memory-to-Memory
    		Read/Write
    		Streaming
    Video input : 0 (loopback: ok)
    Video output: 0 (loopback in)
    Format Video Capture:
    	Width/Height  : 850/565
    	Pixel Format  : 'YUYV'
    	Field         : None
    	Bytes per Line: 1700
    	Size Image    : 960500
    	Colorspace    : SRGB
    Format Video Output:
    	Width/Height  : 850/565
    	Pixel Format  : 'YUYV'
    	Field         : None
    	Bytes per Line: 1700
    	Size Image    : 960500
    	Colorspace    : SRGB
    Streaming Parameters Video Capture:
    	Frames per second: 30.000 (30/1)
    	Read buffers     : 8
    Streaming Parameters Video Output:
    	Frames per second: 30.000 (30/1)
    	Write buffers    : 8
    
    User Controls
    
                        keep_format (bool)   : default=0 value=1
                  sustain_framerate (bool)   : default=0 value=1
                            timeout (int)    : min=0 max=100000 step=1 default=0 value=0
                   timeout_image_io (bool)   : default=0 value=0
                        keep_format (bool)   : default=0 value=1
                  sustain_framerate (bool)   : default=0 value=1
                            timeout (int)    : min=0 max=100000 step=1 default=0 value=0
                   timeout_image_io (bool)   : default=0 value=0

What's the background of this?
I'm trying to run a face detection on my webcam while still having skype use the video. Skype (with v4l1compat) will see the loopback device but only read from it if I do the v4l2loopback-ctl in the dump above. I know that because it did work at some point.

So, what could be the cause of this behaviour?

umlaeute, 2015-07-16:
- which version of *v4l2loopback* are you using?
- which versin of *GStreamer*?
- which version of *skype*?

jcaesar, 2015-07-17:
v4l2loopback: 0.9.1
GStreamer: 0.10.36 (but then again, I don't really care about GStreamer)
Skype: 4.3.0.37
debian/stretch, btw. (could have said that earlier…)

Also, should v4l2loopback-ctl have the same version as v4l2loopback?

    $ v4l2loopback-ctl -v
    /usr/bin/v4l2loopback-ctl 0.6

jcaesar, 2015-07-20:
I was bored, so I tried installing the version from the repo. Same result.

bobismijnnaam, 2016-09-22:
Any progress on this? I have the same problem with `gst-launch-0.10 videotestsrc ! queue ! v4l2sink device=/dev/video2`, where video2 is a v4l2loopback device.

dedenker, 2016-10-07:
Any progress on this?
I am getting the same.

RGD2, 2017-01-24:
Me too - v4l2loopback appears broken on my system.

I'm using Ubuntu 16.04.1 LTS, 
GStreamer Core Library version 0.10.36, 
And v4l2loopback from this repository, built against 4.4.0-59-generic #80-Ubuntu SMP Fri Jan 6 17:47:47 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

```
$ gst-launch-0.10 videotestsrc ! v4l2sink device=/dev/video0
```

dies with:

```
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Failed to query attributes of input 0 in device /dev/video0
Additional debug info:
v4l2_calls.c(134): gst_v4l2_fill_lists (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Failed to get 0 in input enumeration for /dev/video0. (25 - Inappropriate ioctl for device)
Setting pipeline to NULL ...
Freeing pipeline ...
```

umlaeute, 2017-01-24:
@RGD2 please include the version of `v4l2loopback` you are using.

in any case; `gstreamer-0.10` is dead and you should switch to `GStreamer-1.0` (which requires `v4l2loobpack>=0.10`.

i'm inclined to close this as out-of-date.

emmanuelgeoffray, 2017-02-28:
@umlaeute [wiki](https://github.com/umlaeute/v4l2loopback/wiki/Gstreamer) says it is better to use `gstreamer-0.10`. 
Is it better to use `GStreamer-1.0 ` now?

jcaesar, 2017-03-07:
I've tried to reproduce this, but I don't have the installation anymore and I don't get the pausing error in a new one. (This doesn't mean that skype will play the loopback now, but admittedly I've given up on that…)

umlaeute, 2017-03-07:
@emmanuelgeoffray thanks for the reminder, i've updated the wiki (last [week](https://github.com/umlaeute/v4l2loopback/wiki/Gstreamer), but forgot to tell) to only mention `GStreamer-1.0`

closed 2017-03-07 by umlaeute