v4l2loopback/v4l2loopback #94: Raspberry 2 build

sylvainblot, 2015-08-11:
Hello guys,

I'm trying to use v4l2loopback on raspi2, whether I try on the default kernel or revert to 3.18 as suggested here: https://github.com/umlaeute/v4l2loopback/issues/84 i get :

```
pi@divingcam ~/v4l2loopback-0.9.1 $ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback-0.9.1 modules
make[1]: Entering directory '/usr/src/linux-headers-3.18.0-trunk-rpi2'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
  CC [M]  /home/pi/v4l2loopback-0.9.1/v4l2loopback.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/pi/v4l2loopback-0.9.1/v4l2loopback.mod.o
  LD [M]  /home/pi/v4l2loopback-0.9.1/v4l2loopback.ko
make[1]: Leaving directory '/usr/src/linux-headers-3.18.0-trunk-rpi2'

pi@divingcam ~/v4l2loopback-0.9.1 $ sudo insmod v4l2loopback.ko
Error: could not insert module v4l2loopback.ko: Unknown symbol in module

pi@divingcam ~/v4l2loopback-0.9.1 $ uname -a
Linux divingcam 3.18.0-trunk-rpi2 #1 SMP PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv7l GNU/Linux
```

I've the same error with HEAD.

Here is the trace:
```
pi@divingcam ~/v4l2loopback-0.9.1 $ sudo strace insmod v4l2loopback.ko
execve("/sbin/insmod", ["insmod", "v4l2loopback.ko"], [/* 15 vars */]) = 0
brk(0)                                  = 0x7762f000
uname({sys="Linux", node="divingcam", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76fdd000
access("/etc/ld.so.preload", R_OK)      = 0
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=44, ...}) = 0
mmap2(NULL, 44, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76fdc000
close(3)                                = 0
open("/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\270\4\0\0004\0\0\0"..., 512) = 512
lseek(3, 7276, SEEK_SET)                = 7276
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
lseek(3, 7001, SEEK_SET)                = 7001
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0755, st_size=10170, ...}) = 0
mmap2(NULL, 39740, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76fd2000
mprotect(0x76fd4000, 28672, PROT_NONE)  = 0
mmap2(0x76fdb000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1) = 0x76fdb000
close(3)                                = 0
munmap(0x76fdc000, 44)                  = 0
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=64087, ...}) = 0
mmap2(NULL, 64087, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76fa0000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libkmod.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240$\0\0004\0\0\0"..., 512) = 512
lseek(3, 78528, SEEK_SET)               = 78528
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 78232, SEEK_SET)               = 78232
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=79648, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76fdc000
mmap2(NULL, 111004, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76f84000
mprotect(0x76f97000, 28672, PROT_NONE)  = 0
mmap2(0x76f9e000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12) = 0x76f9e000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0`\321\0\0004\0\0\0"..., 512) = 512
lseek(3, 116400, SEEK_SET)              = 116400
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 116072, SEEK_SET)              = 116072
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0644, st_size=117520, ...}) = 0
mmap2(NULL, 181664, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76f57000
mprotect(0x76f74000, 61440, PROT_NONE)  = 0
mmap2(0x76f83000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c) = 0x76f83000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0L\204\1\0004\0\0\0"..., 512) = 512
lseek(3, 1223552, SEEK_SET)             = 1223552
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 2840) = 2840
lseek(3, 1220116, SEEK_SET)             = 1220116
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0755, st_size=1226392, ...}) = 0
mmap2(NULL, 1295768, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76e1a000
mprotect(0x76f42000, 61440, PROT_NONE)  = 0
mmap2(0x76f51000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x127) = 0x76f51000
mmap2(0x76f54000, 9624, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x76f54000
close(3)                                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76fd1000
set_tls(0x76fd14c0, 0x76fd1ba8, 0x76fe0050, 0x76fd14c0, 0x76fe0050) = 0
mprotect(0x76f51000, 8192, PROT_READ)   = 0
mprotect(0x76f9e000, 4096, PROT_READ)   = 0
mprotect(0x76ffa000, 4096, PROT_READ)   = 0
mprotect(0x76fdf000, 4096, PROT_READ)   = 0
munmap(0x76fa0000, 64087)               = 0
brk(0)                                  = 0x7762f000
brk(0x77650000)                         = 0x77650000
uname({sys="Linux", node="divingcam", ...}) = 0
open("/proc/cmdline", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "dma.dmachans=0x7f35 bcm2708_fb.f"..., 4095) = 443
read(3, "", 3652)                       = 0
close(3)                                = 0
getcwd("/home/pi/v4l2loopback-0.9.1", 4096) = 28
stat64("/home/pi/v4l2loopback-0.9.1/v4l2loopback.ko", {st_mode=S_IFREG|0644, st_size=44492, ...}) = 0
open("/home/pi/v4l2loopback-0.9.1/v4l2loopback.ko", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=44492, ...}) = 0
mmap2(NULL, 44492, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76fa5000
init_module(0x76fa5000, 44492, "")      = -1 ENOENT (No such file or directory)
munmap(0x76fa5000, 44492)               = 0
close(3)                                = 0
write(2, "Error: could not insert module v"..., 73Error: could not insert module v4l2loopback.ko: Unknown symbol in module
) = 73
exit_group(1)
```

Thanks !

sylvainblot, 2015-08-11:
I forgot to 
```
make modprobe
```

My bad.

closed 2015-08-11 by sylvainblot

umlaeute, 2015-08-11:
the trick is that `v4l2loopback` needs the `videodev` module to be loaded.

`make modprobe` does that for you

alternatively, when *installing* the module, a call to `depmod` should resolve inter-module dependencies, so when you do `modprobe v4l2loopback`, the system already knows that it also needs to load `videodev`. (that's why the `install` target calls `depmod`)