v4l2loopback/v4l2loopback #99: v4l2loopback_formats.h line 276: error: expected ‘}’ before ‘;’ token

joecool164, 2015-11-04:
There is a bug in v4l2loopback_formats.h:  line 276:

#ifdef V4L2_PIX_FMT_MPEG
{
   .name     = "MPEG-1/2/4 Multiplexed",
     .fourcc   = V4L2_PIX_FMT_MPEG,
     .depth    = 32,
     .flags    = FORMAT_FLAGS_COMPRESSED,
#endif /* V4L2_PIX_FMT_MPEG */
#ifdef V4L2_PIX_FMT_H264
     },
{
   .name     = "H264 with start codes",
     .fourcc   = V4L2_PIX_FMT_H264,
     .depth    = 32,
     .flags    = FORMAT_FLAGS_COMPRESSED,
     },
#endif /* V4L2_PIX_FMT_H264 */

The "}," on line 276 should come before the #endif /* V4L2_PIX_FMT_MPEG */

umlaeute, 2015-11-05:
thanks.

for future reference:
- it would be great if you could provide a link to the actual problematic line rather than pasting some random code-snippet without context. something like https://github.com/umlaeute/v4l2loopback/blob/1d7e2976d9269d476e8f8ee2361506d06a3f79f5/v4l2loopback_formats.h#L276

- even better would be a pull-request with the actual fix :-)

closed 2015-11-05 by umlaeute

joecool164, 2015-11-05:
Thanks.  

Yeah, I'm new to github.  I'll try to add the additional information next time.  Thanks for the pointers and the fix.