v4l2loopback/v4l2loopback #103: Can't read private key

R7914T, 2015-12-09:
Were do I find the private key:

robins@robins-desktop:~/Desktop/v4l2loopback-master$ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/robins/Desktop/v4l2loopback-master modules
make[1]: Entering directory `/usr/src/linux-headers-3.19.0-39-generic'
  CC [M]  /home/robins/Desktop/v4l2loopback-master/v4l2loopback.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/robins/Desktop/v4l2loopback-master/v4l2loopback.mod.o
  LD [M]  /home/robins/Desktop/v4l2loopback-master/v4l2loopback.ko
make[1]: Leaving directory `/usr/src/linux-headers-3.19.0-39-generic'
robins@robins-desktop:~/Desktop/v4l2loopback-master$ sudo make install
[sudo] password for robins: 
make -C /lib/modules/`uname -r`/build M=/home/robins/Desktop/v4l2loopback-master modules_install
make[1]: Entering directory `/usr/src/linux-headers-3.19.0-39-generic'
  INSTALL /home/robins/Desktop/v4l2loopback-master/v4l2loopback.ko
Can't read private key
  DEPMOD  3.19.0-39-generic
make[1]: Leaving directory `/usr/src/linux-headers-3.19.0-39-generic'
depmod -a  `uname -r`
robins@robins-desktop:~/Desktop/v4l2loopback-master$ sudo modprobe v412loopback
modprobe: FATAL: Module v412loopback not found.

kripton, 2015-12-10:
You have a typo there. Try "sudo modprobe v4l2loopback" instead of "sudo modprobe v412loopback"

umlaeute, 2015-12-10:
not an error (unrelated to v4l2loopback, but more of a ubuntu feature for package maintainers; won't break the build).

as @kripton has pointed out, your problem is a simple typo.

closed 2015-12-10 by umlaeute