v4l2loopback/v4l2loopback #104: Does not install on kernel 4.1.13+

chrisf4lc0n, 2015-12-20:
Tried to compile it from source, so installed the linux-headers-4.1.13+_4.1.13+-2_armhf.deb kernel on my Raspberry Pi, then it completed the installation without any visible errors, but did not work spiting:
modprobe: FATAL: Module v4l2loopback not found.

Then I tried the install v4l2loopback-dkms it installed but the same error...
Only the sudo module-assistant auto-install v4l2loopback-source spat out incompatibility with kernel 4.1.13+ kernel errors...

The only solution I have got to the problem at the moment is to revert to older 3.x kernels and try again...

chrisf4lc0n, 2015-12-20:
Tested it with 4.1.15+ kernel and unfortunately it did the same thing:
pi@DoorBell:~ $ sudo cat /var/lib/dkms/v4l2loopback/0.8.0/build/make.log
DKMS make.log for v4l2loopback-0.8.0 for kernel 4.1.15+ (armv6l)
Sun 20 Dec 18:38:33 UTC 2015
Building v4l2-loopback driver...
make -C /lib/modules/4.1.15+/build M=/var/lib/dkms/v4l2loopback/0.8.0/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.1.15+'
  CC [M]  /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c: In function ‘attr_store_maxopeners’:
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:486:2: error: implicit declaration of function ‘strict_strtoul’ [-Werror=implicit-function-declaration]
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c: In function ‘init_vdev’:
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:7: error: ‘struct video_device’ has no member named ‘debug’
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:17: error: ‘V4L2_DEBUG_IOCTL’ undeclared (first use in this function)
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:17: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:36: error: ‘V4L2_DEBUG_IOCTL_ARG’ undeclared (first use in this function)
cc1: some warnings being treated as errors
scripts/Makefile.build:264: recipe for target '/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o' failed
make[2]: *** [/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o] Error 1
Makefile:1384: recipe for target '_module_/var/lib/dkms/v4l2loopback/0.8.0/build' failed
make[1]: *** [_module_/var/lib/dkms/v4l2loopback/0.8.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.1.15+'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2

umlaeute, 2015-12-21:
`v4l2loopback-0.8`? this was released in 2013, long before there was a 4.0-kernel.

the `ChangeLog` file mentions for the `0.9.0` release:

> * Fix for kernel 4.0

so you might have more luck updating your v4l2oopback.

closed 2015-12-21 by umlaeute