v4l2loopback/v4l2loopback #106: Flipping webcam image

cass00, 2016-01-25:
I'm trying to rotate the output of my webcam using v4l2loopback via gst-launch.

The following command works:
```
gst-launch-1.0 -v v4l2src  ! 'video/x-raw,width=640,format=YUY2' ! videoconvert ! videoflip method=clockwise ! xvimagesink
```
However if I'm trying to pipe into a v4l2sink, i.e.
```
gst-launch-1.0 -v v4l2src  ! 'video/x-raw,width=640,format=YUY2' ! videoconvert ! videoflip method=clockwise ! v4l2sink device=/dev/video2
```
I'm getting:
```
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2943): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming task paused, reason error (-5)
Execution ended after 0:00:00.313524448
```

Without the flipping the above command works.

umlaeute, 2016-01-28:
what exactly do you mean with *Without the flipping the above command works.*?

cass00, 2016-01-28:
```gst-launch-1.0 -v v4l2src  ! 'video/x-raw,width=640,format=YUY2' ! videoconvert ! v4l2sink device=/dev/video2```

works

umlaeute, 2016-12-02:
hopefully fixed in `0.10.0`

closed 2016-12-02 by umlaeute

cass00, 2016-12-28:
Can confirm that it works now.