v4l2loopback/v4l2loopback #107: cant install on Rasp2 with K 4.1.13-v7+

nidez, 2016-01-31:
I'm having troubles trying to install v4l2loopback, maybe it's something i'm mistaking... but...

after `root@raspberrypi:~ # sudo apt-get install v4l2loopback-dkms`
done `root@raspberrypi:~ # modprobe v4l2loopback`
and got `modprobe: FATAL: Module v4l2loopback not found.`

then tried to 
`
root@raspberrypi:~ # wget https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/linux-headers-4.1.13-v7%2B_4.1.13-v7%2B-2_armhf.deb
root@raspberrypi:~ # dpkg -i linux-headers-4.1.13-v7+_4.1.13-v7+-2_armhf.deb
`

all ok, no errors. 
downloaded this code `
root@raspberrypi:~ # git clone https://github.com/umlaeute/v4l2loopback
root@raspberrypi:~ # make && make install
` 
all ok no errors.

`root@raspberrypi:~/v4l2loopback # make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/root/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
  CC [M]  /root/v4l2loopback/v4l2loopback.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /root/v4l2loopback/v4l2loopback.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'

root@raspberrypi:~/v4l2loopback # make install
make -C /lib/modules/`uname -r`/build M=/root/v4l2loopback modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
  INSTALL /root/v4l2loopback/v4l2loopback.ko
  DEPMOD  4.1.13-v7
make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'
depmod -a  `uname -r` `

    ..... BUT ....

`
root@raspberrypi:~/v4llpb/v4l2loopback # modprobe v4l2loopback
modprobe: FATAL: Module v4l2loopback not found.
`

What am I missing ?

nidez, 2016-01-31:
update:
after removing v4l2loopback-dkms, autoremoving unused packets, rebooted... i tried to apt-get it again

> root@raspberrypi:/home/pi # apt-get install v4l2loopback-dkms
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> Suggested packages:
>   v4l2loopback-utils
> The following NEW packages will be installed:
>   v4l2loopback-dkms
> 0 upgraded, 1 newly installed, 0 to remove and 91 not upgraded.
> Need to get 0 B/31.2 kB of archives.
> After this operation, 115 kB of additional disk space will be used.
> Selecting previously unselected package v4l2loopback-dkms.
> (Reading database ... 148507 files and directories currently installed.)
> Preparing to unpack .../v4l2loopback-dkms_0.8.0-4_all.deb ...
> Unpacking v4l2loopback-dkms (0.8.0-4) ...
> Setting up v4l2loopback-dkms (0.8.0-4) ...
> Loading new v4l2loopback-0.8.0 DKMS files...
> First Installation: checking all kernels...
> It is likely that 4.1.13-v7+ belongs to a chroot's host
> Building initial module for 4.1.13-v7+
> Error! Bad return status for module build on kernel: 4.1.13-v7+ (armv7l)
> Consult /var/lib/dkms/v4l2loopback/0.8.0/build/make.log for more information.

this is the make log

> root@raspberrypi:/home/pi # cat /var/lib/dkms/v4l2loopback/0.8.0/build/make.log
> DKMS make.log for v4l2loopback-0.8.0 for kernel 4.1.13-v7+ (armv7l)
> Sun 31 Jan 20:42:05 UTC 2016
> Building v4l2-loopback driver...
> make -C /lib/modules/4.1.13-v7+/build M=/var/lib/dkms/v4l2loopback/0.8.0/build modules
> make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
>   CC [M]  /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o
> /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c: In function ‘attr_store_maxopeners’:
> /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:486:2: error: implicit declaration of function ‘strict_strtoul’ [-Werror=implicit-function-declaration]
> /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c: In function ‘init_vdev’:
> /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:7: error: ‘struct video_device’ has no member named ‘debug’
> /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:17: error: ‘V4L2_DEBUG_IOCTL’ undeclared (first use in this function)
> /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:17: note: each undeclared identifier is reported only once for each function it appears in
> /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:36: error: ‘V4L2_DEBUG_IOCTL_ARG’ undeclared (first use in this function)
> cc1: some warnings being treated as errors
> scripts/Makefile.build:264: recipe for target '/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o' failed
> make[2]: *** [/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o] Error 1
> Makefile:1384: recipe for target '_module_/var/lib/dkms/v4l2loopback/0.8.0/build' failed
> make[1]: *** [_module_/var/lib/dkms/v4l2loopback/0.8.0/build] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'
> Makefile:42: recipe for target 'v4l2loopback.ko' failed
> make: *** [v4l2loopback.ko] Error 2

I'm quite noob, and i really can't understand it...

nidez, 2016-01-31:
and last update i can give:

root@raspberrypi:/lib/modules # uname -r
> 4.1.13-v7+

root@raspberrypi:/lib/modules # ls -al
> total 24
> drwxr-xr-x  6 root root 4096 Jan 31 19:48 .
> drwxr-xr-x 19 root root 4096 Nov 21 20:48 ..
> drwxr-xr-x  3 root root 4096 Jan 31 17:20 3.6-trunk-rpi
> drwxr-xr-x  3 root root 4096 Jan 31 20:58 4.1.13+
> drwxr-xr-x  3 root root 4096 Jan 31 19:48 4.1.13-v7
> drwxr-xr-x  3 root root 4096 Jan 31 20:21 4.1.13-v7+

v4l2loopback.ko is present inside /lib/modules/4.1.13-v7/extra/
but while the others kernel folders have a lot of stuff in it (and the "kernel" folder too) the "4.1.13-v7" has only the "extra" folder inside

root@raspberrypi:/lib/modules/4.1.13-v7 # ls -al
> total 12K
> drwxr-xr-x 3 root root 4.0K Jan 31 19:48 .
> drwxr-xr-x 6 root root 4.0K Jan 31 19:48 ..
> drwxr-xr-x 2 root root 4.0K Jan 31 19:48 extra

root@raspberrypi:/lib/modules/4.1.13-v7+ # ls -al
> total 1.6M
> drwxr-xr-x  3 root root 4.0K Jan 31 20:21 .
> drwxr-xr-x  6 root root 4.0K Jan 31 19:48 ..
> lrwxrwxrwx  1 root root   33 Dec  9 22:06 build -> /usr/src/linux-headers-4.1.13-v7+
> drwxr-xr-x 12 root root 4.0K Jan 31 20:56 kernel
> -rw-r--r--  1 root root 420K Jan 31 20:21 modules.alias
> -rw-r--r--  1 root root 435K Jan 31 20:21 modules.alias.bin
> -rw-r--r--  1 root root 4.3K Nov 18 16:01 modules.builtin
> -rw-r--r--  1 root root 5.8K Jan 31 20:21 modules.builtin.bin
> -rw-r--r--  1 root root 131K Jan 31 20:21 modules.dep
> -rw-r--r--  1 root root 193K Jan 31 20:21 modules.dep.bin
> -rw-r--r--  1 root root  302 Jan 31 20:21 modules.devname
> -rw-r--r--  1 root root  56K Nov 18 16:01 modules.order
> -rw-r--r--  1 root root   55 Jan 31 20:21 modules.softdep
> -rw-r--r--  1 root root 147K Jan 31 20:21 modules.symbols
> -rw-r--r--  1 root root 185K Jan 31 20:21 modules.symbols.bin

root@raspberrypi:/lib/modules/4.1.13+ # ls -al
> total 1612
> drwxr-xr-x  3 root root   4096 Jan 31 20:58 .
> drwxr-xr-x  6 root root   4096 Jan 31 19:48 ..
> drwxr-xr-x 12 root root   4096 Jan 31 20:58 kernel
> -rw-r--r--  1 root root 429070 Nov 18 16:01 modules.alias
> -rw-r--r--  1 root root 444522 Nov 18 16:01 modules.alias.bin
> -rw-r--r--  1 root root   4341 Nov 18 16:01 modules.builtin
> -rw-r--r--  1 root root   5903 Nov 18 16:01 modules.builtin.bin
> -rw-r--r--  1 root root 133376 Nov 18 16:01 modules.dep
> -rw-r--r--  1 root root 196892 Nov 18 16:01 modules.dep.bin
> -rw-r--r--  1 root root    302 Nov 18 16:01 modules.devname
> -rw-r--r--  1 root root  56337 Nov 18 16:01 modules.order
> -rw-r--r--  1 root root     55 Nov 18 16:01 modules.softdep
> -rw-r--r--  1 root root 149335 Nov 18 16:01 modules.symbols
> -rw-r--r--  1 root root 188435 Nov 18 16:01 modules.symbols.bin

I've tried to copy the "extra" folder (containing only v4l2loopback.so) on the others K folders but with no other result than usual 

root@raspberrypi:/lib/modules # dpkg-reconfigure v4l2loopback-dkms
> ------------------------------
> Deleting module version: 0.8.0
> completely from the DKMS tree.
> ------------------------------
> Done.
> Loading new v4l2loopback-0.8.0 DKMS files...
> It is likely that 4.1.13-v7+ belongs to a chroot's host
> Building initial module for 4.1.13-v7+
> Error! Bad return status for module build on kernel: 4.1.13-v7+ (armv7l)
> Consult /var/lib/dkms/v4l2loopback/0.8.0/build/make.log for more information.

umlaeute, 2016-02-02:
- `v0.8.0` does *not* support `linux-4.x*`, you will need *at least* `v0.9.0`

closed 2016-02-02 by umlaeute

nidez, 2016-02-02:
v 0.9.2    --- > long story, short : still not able to install...  
YES I'm NOOB, ok, fine, I'm learning, I would love to RTFM if I only was able to find it ! 
Could you please point me in the right direction ?

- fresh new raspbian image
- downloaded 0.9.2, unzipped
- installed 4.1.13-v7+ headers (from .deb), with gcc-4.7, linked to build/ ecc.. ecc...
- root@raspberrypi:/usr/src/v4l2loopback # make       ... all good...

> Building v4l2-loopback driver...
> make -C /lib/modules/`uname -r`/build M=/usr/src/v4l2loopback modules
> make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
>   Building modules, stage 2.
>   MODPOST 1 modules
> make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'

- root@raspberrypi:/usr/src/v4l2loopback # make install    ... not good....

> make -C /lib/modules/`uname -r`/build M=/usr/src/v4l2loopback modules_install
> make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
>   INSTALL /usr/src/v4l2loopback/v4l2loopback.ko
>   DEPMOD  4.1.13-v7
> make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'
> depmod -a  `uname -r`
**> depmod: WARNING: could not open /lib/modules/4.1.13-v7+/modules.order: No such file or directory
> depmod: WARNING: could not open /lib/modules/4.1.13-v7+/modules.builtin: No such file or directory**
> 

root@raspberrypi:/usr/src/v4l2loopback # modprobe v4l2loopback

**> modprobe: ERROR: ../libkmod/libkmod.c:508 kmod_lookup_alias_from_builtin_file() could not open builtin file '/lib/modules/4.1.13-v7+/modules.builtin.bin'
> modprobe: FATAL: Module v4l2loopback not found.**