v4l2loopback/v4l2loopback #108: V.0.9.1 cant install on Rasp2 with K 4.1.13-v7+  neither..

nidez, 2016-02-03:
v 0.9.1   --- > long story, short : still not able to install...  
YES I'm NOOB, ok I'm learning, I would love to RTFM if I only was able to find it ! 
Could you please point me in the right direction ? Pls help this is getting very frustrating..

- fresh new raspbian image
- downloaded 0.9.2, unzipped
- installed 4.1.13-v7+ headers (from .deb), with gcc-4.7, linked to build/ ecc.. ecc...
- `root@raspberrypi:/usr/src/v4l2loopback # make   `    ... all good...

> Building v4l2-loopback driver...
> make -C /lib/modules/`uname -r`/build M=/usr/src/v4l2loopback modules
> make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
>   Building modules, stage 2.
>   MODPOST 1 modules
> make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'

- `root@raspberrypi:/usr/src/v4l2loopback # make install  `  ... not good....

> make -C /lib/modules/`uname -r`/build M=/usr/src/v4l2loopback modules_install
> make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
>   INSTALL /usr/src/v4l2loopback/v4l2loopback.ko
>   DEPMOD  4.1.13-v7
> make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'
> depmod -a  `uname -r`
**> depmod: WARNING: could not open /lib/modules/4.1.13-v7+/modules.order: No such file or directory
> depmod: WARNING: could not open /lib/modules/4.1.13-v7+/modules.builtin: No such file or directory**

- `root@raspberrypi:/usr/src/v4l2loopback # modprobe v4l2loopback`

> **modprobe: ERROR: ../libkmod/libkmod.c:508 kmod_lookup_alias_from_builtin_file() could not open builtin file '/lib/modules/4.1.13-v7+/modules.builtin.bin'
> modprobe: FATAL: Module v4l2loopback not found.**

- touched those files and make install again
`root@raspberrypi:/usr/src/v4l2loopback # touch /lib/modules/4.1.13-v7+/modules.order
root@raspberrypi:/usr/src/v4l2loopback # touch /lib/modules/4.1.13-v7+/modules.builtin
root@raspberrypi:/usr/src/v4l2loopback # make install`

> make -C /lib/modules/`uname -r`/build M=/usr/src/v4l2loopback modules_install
> make[1]: Entering directory '/usr/src/linux-headers-4.1.13-v7+'
>   DEPMOD  4.1.13-v7
> make[1]: Leaving directory '/usr/src/linux-headers-4.1.13-v7+'
> depmod -a  `uname -r`

- but still no luck `root@raspberrypi:/usr/src/v4l2loopback # modprobe v4l2loopback`

> modprobe: FATAL: Module v4l2loopback not found.
 
- realized that the v4l2loopback.ko file is located in /lib/modules/4.1.13-v7/extra while i'm on ...-v7+/ so I copied the "extra" folder around in a couple of places (don't know which is the right one)
`root@raspberrypi:/lib/modules/4.1.13-v7 # mv extra/ ../4.1.13-v7+/
root@raspberrypi:/lib/modules # cd ../4.1.13-v7+/
root@raspberrypi:/lib/modules/4.1.13-v7+ # cp -r extra build/
root@raspberrypi:/lib/modules/4.1.13-v7+ # depmod -a  `uname -r`
root@raspberrypi:/lib/modules/4.1.13-v7+ # modprobe v4l2loopback`

> modprobe: ERROR: could not insert 'v4l2loopback': Unknown symbol in module, or unknown parameter (see dmesg)

- `root@raspberrypi:/lib/modules/4.1.13-v7+ # dmesg`

> [  144.516503] v4l2loopback: Unknown symbol video_ioctl2 (err 0)
> [  144.516560] v4l2loopback: Unknown symbol v4l2_ctrl_handler_init_class (err 0)
> [  144.516650] v4l2loopback: Unknown symbol video_devdata (err 0)
> [  144.516685] v4l2loopback: Unknown symbol v4l2_ctrl_new_custom (err 0)
> [  144.516713] v4l2loopback: Unknown symbol video_unregister_device (err 0)
> [  144.516746] v4l2loopback: Unknown symbol video_device_alloc (err 0)
> [  144.516791] v4l2loopback: Unknown symbol v4l2_device_register (err 0)
> [  144.516836] v4l2loopback: Unknown symbol __video_register_device (err 0)
> [  144.516868] v4l2loopback: Unknown symbol v4l2_ctrl_handler_free (err 0)
> [  144.516884] v4l2loopback: Unknown symbol v4l2_device_unregister (err 0)
> [  144.516900] v4l2loopback: Unknown symbol video_device_release (err 0)

umlaeute, 2016-02-04:
i'm pretty sure your problem is not related to `v4l2loopback` but rather a general issue with building modules for your kernel (according to #107 you have basically downloaded a random set of kernel-headers from some random place; but kernel-headers need to match the used kernel *exactly* - a version string is just a hint, no guarantee...).

anyhow: `v4l2loopback` requires the `videodev` module to be loaded. (that's what `depmod -a` is supposed to help at).
because you have a kernel/header mismatch, `depmod -a` did not do it's job. but you can load the `videodev` manually *before* trying to load `v4l2loopback`:

    modprobe videodev
    make modprobe

(the last line will try load the just-built `v4l2loopback.ko` module, without requiring it to be installed; if the module is installed in a proper place, you can of course also run `modprobe v4l2loopback`)

patrickpoirier51, 2016-02-12:
I tried loading it apt-get install v4l2loopback-utils on a Raspberrypi 2 loaded with latest Raspbian Jessie image (feb 9 2016) that runs on a 4.1.x Kernel.
The installation failed with '' It is likely that 4.1.7-v7+ belongs to a chroot's host''

I rally want to use this latest kernel (it has the openGL drivers working) with v4l2loopback ...please help

umlaeute, 2016-02-12:
@patrickpoirier51  this is really an issue of your kernel/headers, and unrelated to v4l2loopback.

please report the bug with the raspbian developers (they should start shipping kernel-headers for their kernels)

closed 2016-02-12 by umlaeute

patrickpoirier51, 2016-02-12:
Ok good I will report that , thanks for your fas answer.
Hopefully they will fix it rapidly so you wont be submerge with the same request

madovsky, 2016-02-12:
I use v4l2loopback with my own compiled kernel 4.4.1 and works flawlessly

umlaeute, 2016-02-12:
@madovsky  which only confirms that the issue is only with the installed kernel+header (and outside the scope of v4l2loopback)

patrickpoirier51, 2016-02-12:
Great , I'll try to compile it then, do you have any special procedures to suggest ?
BTW I left the kernel-header stuff on RPI forum
Regards

madovsky, 2016-02-12:
nothing special, just respect what rasp needs

nidez, 2016-02-12:
i finally managed to get it work,
- installed K H from .deb (downloaded from https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/ )
- downloaded v4l2loopback source version 0.9.0 .... 0.9.1 didnt work, and 0.8 neither. (https://github.com/umlaeute/v4l2loopback/releases)
- compiled just fine.

@umlaeute  i think it's better to update the readme file on the Kernel and distributions topics, from:

(kernel) 
<= 2.6.27 definitely will NOT work
2.6.28 - 2.6.31 may work (seems to compile but i cannot test)
>= 2.6.32 should work
>= 3.0.0 should work as well

to 

<= 2.6.27 definitely will NOT work
2.6.28 - 2.6.31 may work (seems to compile but i cannot test)
>= 2.6.32 should work
>= 3.0.0 should work as well until 4.x
**for 4.x use version 0.9.0 at least !**

and from

(distributions)
This means, that it is also part of Debian-derived distributions, including Ubuntu (starting with natty). The most convenient way is to install the package "v4l2loopback-dkms":
# aptitude install v4l2loopback-dkms

to

This means, that it is also part of Debian-derived distributions, including Ubuntu (starting with natty). The most convenient way is to install the package "v4l2loopback-dkms":
# aptitude install v4l2loopback-dkms
**(warning: on raspberry raspibian jessie do not use the aptitude packet, download 0.9.0 from source.)**

bye !
;)

umlaeute, 2016-02-12:
@nidez  thanks for goind through all this.

i'm rather puzzled about the `0.9.0` vs `0.9.1`  thing: the *only* difference between the two is that all the iles have been updated to say `0.9.1` instead of `0.8.0` [sic!]

nidez, 2016-02-12:
really dunno why, well i'm a noob in compiling stuff, i'm not a dev at all... maybe mine is not a real solution, maybe it's just a lucky combination of factors.

patrickpoirier51, 2016-02-19:
I do have difficulties to get the packages (4.1.13 or 4.1.17) compiled.
Build fresh image and apt-get update
Download from https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/  with uname-a version
sudo dpkg -i linux-headers, unpack then stops requesting gcc4.7 that I have to apt-get -f install
then it end with dkms not found.
The when I reboot It start sequence but hangs without GUI screen.
I can access remotely with terminal,

I guess it is missing tools and other stuff ?!?!

patrickpoirier51, 2016-02-20:
OK I made it work, on kernel 4.1.17 with release 9.1
here is the requirements:
sudo apt-get update
sudo apt-get install dkms build-essential

check gcc --version  if not ==4.7.3
sudo apt-get install gcc-4.7
sudo rm /usr/bin/gcc
sudo ln -s /usr/bin/gcc-4.7 /usr/bin/gcc

this is how i load it:
sudo modprobe videodev
sudo insmod ./v4l2loopback.ko