v4l2loopback/v4l2loopback #113: cant compile the kernal modle

gemad, 2016-04-22:
after downloading the v4l2loopback am trying to run `make` but am getting this error 

`make -C /lib/modules/`uname -r`/build M=/home/gemad/Desktop/localization /v4l2loopback-master modules
make[1]: Entering directory `/usr/src/linux-headers-3.19.0-51-generic'
arch/x86/Makefile:129: CONFIG_X86_X32 enabled but no binutils support
Makefile:669: Cannot use CONFIG_CC_STACKPROTECTOR_REGULAR: -fstack-protector not supported by compiler
make[1]: *** No rule to make target `/v4l2loopback-master'.  Stop.
make[1]: Leaving directory `/usr/src/linux-headers-3.19.0-51-generic'
make: *** [v4l2loopback.ko] Error 2
`

my kernal is `3.19.0-51-generic`

and i download the kernal header using those commands 
`sudo apt-get install linux-headers-$(uname -r)`
`sudo apt-get update && sudo apt-get upgrade`

what should i do ?

umlaeute, 2016-04-23:
try installing the `v4l2loopback-dkms` package instead.

if that works, you should also be able to build the kernel module yourself (since the dkms package also builds it)

gemad, 2016-04-23:
@umlaeute 
i tried to use those 
`sudo apt-get install linux-generic `

the output : `linux-generic is already the newest version`.

`sudo apt-get install v4l2loopback-dkms`

the output : `linux-generic is already the newest version`.

`sudo modprobe v4l2loopback`

the output : `modprobe: FATAL: Module v4l2loopback not found`.

umlaeute, 2016-04-23:
how do you relate the output `linux-generic is already the newest version` with the input `sudo apt-get install v4l2loopback-dkms`?

in any case, if your distribution (which one?) *has* a `v4l2loopback-dkms` module and it fails to install, you should report a bug with your distribution.

gemad, 2016-04-23:
@umlaeute sorry i edited it , do you mean the kernal , mine 
`uname -r`    `3.19.0-51-generic`

umlaeute, 2016-04-23:
no. i mean distribution. the following might help (or not):

    lsb_release -a

gemad, 2016-04-23:
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.4 LTS
Release:	14.04
Codename:	trusty

note that there is no problem with a pc have the same redistribution and different kernal 
it works for a 3.19.0-41 but not mine don know why

umlaeute, 2016-04-23:
so what's `pwd`?

gemad, 2016-04-23:
/home/gemad

umlaeute, 2016-04-23:
are you trying to  tell me that you have a file `/home/gemad/v4l2loopback.c`?
> and while you are there, what's the output of pwd when run from the v4l2loopback-directory?

umlaeute, 2016-04-23:
also, please provide context in your answers. e.g. rather than saying

> /home/gemad

do something like:

> $ pwd
> /home/gemad

gemad, 2016-04-23:
okii i got your point , 

```
$pwd
/home/gemad/Desktop/localization /v4l2loopback-master
```

the problem was the space in the localization folder name now it's compiled , i checked this more than time to make sure there is no space in the name of the folder , but its the first time to notice that , thanks anyway

umlaeute, 2016-04-23:
great that it finally worked.

actually, the space issue pops up as the first page of results when googling for [CONFIG_X86_X32 enabled but no binutils support](https://www.google.at/search?q=CONFIG_X86_X32+enabled+but+no+binutils+support)

closed 2016-04-23 by umlaeute

gemad, 2016-04-23:
i know , but never notice it until i paste it in this write  box , thanks