v4l2loopback/v4l2loopback #116: Cannot use v4l2loopback device as sink for gstreamer1.0 (Internal data flow error)

juanitobananas, 2016-06-29:
First of all, thank you very much for this great piece of software!! :smile: 

I have a small issue with a v4l2loopback device. I am trying to use it with gstreamer (1.0) to no avail. At least not with the options I'd like to use. I plan to send a stream through the LAN to another computer and to test if the creation of the device was working, first of all I tested this:

```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! v4l2sink device=/dev/video1
```

and it worked. I can see the stream using vlc :smiley: 

As I want to send it through LAN, I want to compress it (at least a little) and tried with mjpeg:

```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! jpegenc ! jpegdec ! v4l2sink device=/dev/video1
```

but unfortunately this didn't work (I have tried 'queuing ' and 'videoconverting' everything and it didn't help). By the way, sending the stream to autovideosink does work, so I guess the encoding and decoding *do* work.

I see an `Internal data flow error`. Here you can see all the output: http://paste.scratchbook.ch/view/ffa6a77c

I spoke to the gstreamer devs and thanks to them I managed to execute it in a *very* verbose way:
http://paste.scratchbook.ch/view/a2fd58b8

And saw that `No free buffer found in the pool at index 1.` Unfortunately I have no idea how this all works, but apparently this might have to do with `the driver returning random buffer indexes`. Apparently, the queue gets bigger - which isn't necessarily a bad thing - when you use more gst elements and v4l2loopback doesn't negotiate the queue size correctly (this is all something I 'learned' yesterday while speaking to the devs on irc, so please forgive me if it is completely inaccurate).

Could this be? Do you have any idea why this is happening? Am I doing something wrong?

It would be great if you could give me a little hand with the issue if you find the time.

Once again, thank you very much!

juanitobananas, 2016-06-29:
It might also be related with #112 so here is the recommended helpful output of gstreamer `GST_DEBUG=*v4l2allocator*:9 gst-launch-1.0 ...` in case it is of any help:

http://paste.scratchbook.ch/view/98aa4434

jvanatta, 2016-11-04:
@juanitobananas I think I'm seeing the same issue. Directly copying to /dev/video1 works, but doing any sort of processing or using a filesrc results in an internal data flow error. Did you have luck on this?

juanitobananas, 2016-11-15:
@jvanatta: Yup, that sounds like the problem I have. Sorry, I am afraid I gave up.

jvanatta, 2016-11-15:
@juanitobananas I was able to get things working by using a tee to force a memory copy in the gstreamer pipeline, as suggested by ndufresne. See the end of this issue thread: https://github.com/umlaeute/v4l2loopback/issues/83

As an example, here's my one-liner to play a video through v4l2 as /dev/video0:

gst-launch-1.0 filesrc location=$FILENAME ! "video/x-raw, format=YUY2, interlace-mode=interleaved" ! decodebin ! videoconvert !  "video/x-raw, format=YUY2, interlace-mode=progressive" ! tee ! v4l2sink

The  ! tee ! pad at the end made all the difference.

juanitobananas, 2016-11-17:
@jvanatta: Cool, thank you very much for that!

silencefall, 2016-12-01:
Hi
I also met this issue， and modify v4l2loopback.c as below.  It seems fixed.
Environment: ubuntu 14.04, gstreamer 10.1
Test pipeline:
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video1
gst-launch-1.0 v4l2src device=/dev/video1 ! glimagesink

Patch:
diff --git a/v4l2loopback.c b/v4l2loopback.c
index 3fe6ee6..103f695 100644
--- a/v4l2loopback.c
+++ b/v4l2loopback.c
@@ -1619,7 +1619,7 @@ static int vidioc_dqbuf(struct file *file, void *private_data, struct v4l2_buffe
                *buf = dev->buffers[index].buffer;
                return 0;
        case V4L2_BUF_TYPE_VIDEO_OUTPUT:
-               b = list_entry(dev->outbufs_list.next, struct v4l2l_buffer, list_head);
+               b = list_entry(dev->outbufs_list.prev, struct v4l2l_buffer, list_head);
                list_move_tail(&b->list_head, &dev->outbufs_list);
                dprintkrw("output DQBUF index: %d\n", b->buffer.index);
                unset_flags(b);

silencefall, 2016-12-01:
Hi
I also met this issue. It seems fixed with this patch
https://github.com/silencefall/v4l2loopback/commit/97dc86bcddd74cc97254bbca0cebbd55569c3cc6
Test environment:
ubuntu 14.04, gstreamer 10.1
Test pipeline:
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video1
gst-launch-1.0 v4l2src device=/dev/video1 ! glimagesink

umlaeute, 2016-12-02:
hopefully fixed in `0.10.0`

closed 2016-12-02 by umlaeute