v4l2loopback/v4l2loopback #117: cross compilation for arm - nanopi M3 board

nccchirag, 2016-07-06:
Kindly can you please guide on cross compiling this module for ARM architecture. 

Below is what I have gotten thru

1. modified KERNEL_DIR path to point to crosscompiled kernel built for the embedded board
a. compiled .ko file for the board ( Nanopi M3 ) with their toolchain on host machine
2. copied the .ko file on the drivers directory on embedded board
3. depmod -a 
4. modprobe v4l2loopback video_nr=10 FAILED with no error

Cannot create a new video device !

Kindly guide

nccchirag, 2016-07-09:
got it installed

closed 2016-07-09 by nccchirag

umlaeute, 2016-07-11:
so what solved the problem?

nccchirag, 2016-07-12:
I had to compile both on the host machine and the target board
Also, modified the default KERNEL directory path in the Makefile to point
to the compiled kernel from the host machine (Ubuntu x64)

Thank You
Best Regards,
Chirag A.