v4l2loopback/v4l2loopback #119: Can not Build on 4.4.13-5-osmc

tkilaaa, 2016-08-17:
If i tried this:
```
$ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/osmc/v4l2loopback modules
make[1]: Entering directory '/lib/modules/4.4.13-5-osmc/build'
  CC [M]  /home/osmc/v4l2loopback/v4l2loopback.o
In file included from ./arch/arm/include/asm/page.h:163:0,
                 from ./arch/arm/include/asm/thread_info.h:17,
                 from include/linux/thread_info.h:54,
                 from include/asm-generic/preempt.h:4,
                 from arch/arm/include/generated/asm/preempt.h:1,
                 from include/linux/preempt.h:59,
                 from include/linux/spinlock.h:50,
                 from include/linux/vmalloc.h:4,
                 from /home/osmc/v4l2loopback/v4l2loopback.c:17:
./arch/arm/include/asm/memory.h:22:25: fatal error: mach/memory.h: No such file or directory
 #include <mach/memory.h>
                         ^
compilation terminated.
scripts/Makefile.build:264: recipe for target '/home/osmc/v4l2loopback/v4l2loopback.o' failed
make[2]: *** [/home/osmc/v4l2loopback/v4l2loopback.o] Error 1
Makefile:1385: recipe for target '_module_/home/osmc/v4l2loopback' failed
make[1]: *** [_module_/home/osmc/v4l2loopback] Error 2
make[1]: Leaving directory '/lib/modules/4.4.13-5-osmc/build'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
```
The system is up2date, headers loaded...

Best regards
tkilaaa

tkilaaa, 2016-08-17:
Hello again,
after i have download also the source files and copy the missing mach headers, i've got the next problem.
```
$ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/osmc/v4l2loopback modules
make[1]: Entering directory '/lib/modules/4.4.13-5-osmc/build'
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/osmc/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /home/osmc/v4l2loopback/v4l2loopback.ko
make[1]: Leaving directory '/lib/modules/4.4.13-5-osmc/build'

$ su

# make install
make -C /lib/modules/`uname -r`/build M=/home/osmc/v4l2loopback modules_install
make[1]: Entering directory '/lib/modules/4.4.13-5-osmc/build'
  INSTALL /home/osmc/v4l2loopback/v4l2loopback.ko
  DEPMOD  4.4.13-5-osmc
make[1]: Leaving directory '/lib/modules/4.4.13-5-osmc/build'
depmod -a  `uname -r`
depmod: ERROR: fstatat(7, asm-arm): Too many levels of symbolic links
```

Best regards
tkilaaa

umlaeute, 2016-08-22:
afaics these are problems with a (half)broken setup (some recursive symlink tree on your filesystem).

most likely this is an issue of your kernel (whoever provided it), and i fear nothing can be done on the v4l2loopback side.

closed 2016-08-22 by umlaeute