v4l2loopback/v4l2loopback #120: Gstreamer 1.0 (buffer 2 was not queued, this indicate a driver bug)

mariannasb, 2016-09-06:
When I try to run a simple pipeline:
`gst-launch-1.0 --gst-debug=*:3 videotestsrc ! v4l2sink device=/dev/video0`

I get:
`Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.022305954 13092       0x8e4630 ERROR          v4l2allocator gstv4l2allocator.c:1299:gst_v4l2_allocator_dqbuf:<v4l2sink0:pool:sink:allocator> buffer 2 was not queued, this indicate a driver bug.
0:00:00.022350661 13092       0x8e4630 WARN                 basesrc gstbasesrc.c:2948:gst_base_src_loop:<videotestsrc0> error: Internal data flow error.
0:00:00.022369263 13092       0x8e4630 WARN                 basesrc gstbasesrc.c:2948:gst_base_src_loop:<videotestsrc0> error: streaming task paused, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2948): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming task paused, reason error (-5)
`

I also get this message a lot, for a lot of different formats, but I don't know if it is relevant...
`0:00:00.018861072 13092       0x8e4630 WARN                    v4l2 gstv4l2object.c:2389:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at YUYV@48x32: 0
`

And I don't know if this is related to any of the other gstreamer bugs...

I have loaded the module by:
`modprobe v4l2loopback exclusive_caps=0`

dedenker, 2016-10-10:
I am getting the same issue.
Is there progress on this of information I should share?

umlaeute, 2016-12-02:
hopefully fixed in `0.10.0`

closed 2016-12-02 by umlaeute

TheSalarKhan, 2018-03-19:
I'm getting a variant of this as an error.
This is my producer:
```
 gst-launch-1.0 -v v4l2src device=/dev/video0 ! "video/x-raw, format=(string)YV12, width=(int)640, height=(int)480" ! v4l2sink device=/dev/video1
```

This is the consumer and it is crashing with a similar error:
```
gst-launch-1.0 -v v4l2src device=/dev/video1 ! xvimagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-raw, format=(string)YV12, width=(int)640, height=(int)480, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1
/GstPipeline:pipeline0/GstXvImageSink:xvimagesink0.GstPad:sink: caps = video/x-raw, format=(string)YV12, width=(int)640, height=(int)480, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1
0:00:00.077648631  9202 0x55cdcfddb940 ERROR          v4l2allocator gstv4l2allocator.c:1304:gst_v4l2_allocator_dqbuf:<v4l2src0:pool:src:allocator> buffer 1 was not queued, this indicate a driver bug.
0:00:00.077786554  9202 0x55cdcfddb940 WARN                 v4l2src gstv4l2src.c:866:gst_v4l2src_create:<v4l2src0> error: Failed to allocate a buffer
0:00:00.077968894  9202 0x55cdcfddb940 WARN                 basesrc gstbasesrc.c:2939:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:00:00.077995033  9202 0x55cdcfddb940 WARN                 basesrc gstbasesrc.c:2939:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate a buffer
Additional debug info:
gstv4l2src.c(866): gst_v4l2src_create (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0
Execution ended after 0:00:00.048385890
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:00:00.078456448  9202 0x55cdcfdbe2a0 WARN           v4l2allocator gstv4l2allocator.c:784:gst_v4l2_allocator_stop:<v4l2src0:pool:src:allocator> error releasing buffers buffers: Device or resource busy
Setting pipeline to NULL ...
Freeing pipeline ...
```

darenw, 2018-04-18:
Sometimes problems with v4l2sink go away when I put a 'tee' element right before it. I don't know about v4l2src fixes.  This relates to some funny bug in v4l2sink.