v4l2loopback/v4l2loopback #121: Persistent v4l2 output even without a producer

pybean, 2016-09-09:
v4l2loopback works great with chrome/chromium browsers by enabling the "exclusive_caps" parameter. This module is awesome.

Now I am having an issue that the browser won't be able to reopen the v4l2loopback stream without refreshing the page, if I restart the producer(gstreamer) in the middle of playback. The browser log shows that it found the same v4l2 device and its status is 'playing', but the actual video is completely blank.

I wonder if re-initiating v4l2loopback causing problems for chrome, so I am trying to make the v4l2loopback device output frames no matter there's a producer or not.

I tried to use timeout and timeout_image so hopefully the device still would be displaying the timeout image when the producer is lost for some reason. But then I noticed that the v4l2loopback device won't accept stream anymore if I set timeout values.

Any advice will be appreciated.

mikhailnov, 2016-10-28:
Hello,
I have alo encountered a similar issue. Did you find a solution?
> But then I noticed that the v4l2loopback device won't accept input stream anymore if I set timeout values.
Not, it does accept input frames when writing using ffmpeg and does not by Gtreamer 0.10. But in case of ffmpeg the last frame is overlayed on a new one.

benlemond, 2019-07-15:
Was there ever any movement on this topic, or a workaround?  I'm facing similar issues.

stephematician, 2025-01-31:
This is fixed in #611 using Chromium 132.0.6834.110 snap with the follow terminal commands to start the device and feed it a test stream:

```sh
sudo modprobe v4l2loopback exclusive_caps=1 max_buffers=4 video_nr=10
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video10
# open a new terminal
sudo v4l2loopback-ctl set-timeout-image -t 500 /dev/video10 YOUR-TIMEOUT-IMAGE.JPEG
```

Then open Chromium and test your fake camera e.g. via  https://webcamtests.com/. You can go to the terminal with the gstreamer pipeline, hit <kbd>Ctrl</kbd> + <kbd>C</kbd> - and the timeout image will appear. You can then restart the test stream.

On that particular web page - you'll need to set a timeout of less than 1000ms (I think) - otherwise it will assume the camera has died. You can, of course, get the camera back by selecting "Stop" and "Start" again.

closed 2025-02-06 by umlaeute