v4l2loopback/v4l2loopback #122: SSL error when performing installation

Cooper-Yang, 2016-11-24:
Not sure if this is my system's problem ...

Any help would be appreciated

```
cooper@Cooper-Yang:~/ProgramFiles/v4l2loopback$ sudo make install
make -C /lib/modules/`uname -r`/build M=/home/cooper/ProgramFiles/v4l2loopback modules_install
make[1]: Entering directory `/usr/src/linux-headers-4.4.0-47-generic'
  INSTALL /home/cooper/ProgramFiles/v4l2loopback/v4l2loopback.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:169
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:172
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-47-generic
make[1]: Leaving directory `/usr/src/linux-headers-4.4.0-47-generic'
depmod -a  `uname -r`
```

umlaeute, 2016-11-24:
doesn't sound like a problem to me.

apart from having the word "error" in the build-logs (which is most likely due to *you* not being able to sign kernel-stuff - something that `v4l2loopback` cannot do anything about), do you experience any *real* problems?

Cooper-Yang, 2016-11-24:
Well, the real problem I'm facing have nothing to do with this "error". Anyway, thanks.

closed 2016-11-24 by Cooper-Yang

HefnySco, 2017-09-27:
Hi Cooper-Yang,
solution is [here](https://help.ubuntu.com/community/DKMS) 
```
sudo cp -R . /usr/src/v4l2loopback-1.1
sudo dkms add -m v4l2loopback -v 1.1
sudo dkms build -m v4l2loopback -v 1.1
sudo dkms install -m v4l2loopback -v 1.1
```

u may need to install dkms using
`sudo apt install dkms`