v4l2loopback/v4l2loopback #127: Genymotion and detecting video device

casper, 2017-02-06:
Ubuntu 14.04 3.13.0-35-generic 64-bit

I'm using the Genymotion Android emulator (genymotion-2.8.1_x64.bin), and I wanted a dummy video device for the emulator, so that I can stream video or an image to the emulator through v4l2loopback.

I can see video from /dev/video0 with VLC, but Genymotion does not seem to detect /dev/video0, and only presents me their default option "Dummy Webcam" device.

http://i.imgur.com/d8oFkNN.png

Any idea how I could get Genymotion to think /dev/video0 was a real webcam so it would show up in the Genymotion webcam selection dialog?

umlaeute, 2017-02-06:
thanks for your feedback.

you probably should report this as a bug with genymotion, rather than here.
once you (are they) have found out what the culprit it, *and* you found that the problem is indeed with v4l2loopback, then please open a specific bug-report.

closed 2017-02-06 by umlaeute

casper, 2017-02-06:
I'll just put in an intermediate discovery, in case someone else ends up reading this ticket.:

Genymotion requires an UVC compatible webcam. I'm not 100% sure what that implies, but I assume v4l2loopback doesn't deal with anything related to UVC, so that is probably the reason the video device does not show up in the selection dialog.

dpayet, 2017-07-30:
have you succeeded to have the v4l2device in the selection dialog ? Is there an other way to make it work with genymotion ?
Thanks

casper, 2017-07-30:
@dpayet I was not able to get it to work. I asked Genymotion about it on 9.2.2017.    
Here is their reply:

> Thanks for your interest in Genymotion!
> 
> Genymotion can't use an external video or an image to simulate the camera for now.
> We know this is an important feature, and we plan to add it in the future.
> 
> You'll be notified as soon as this feature has been added to Genymotion!
> 

Have not heard anything back from them since then. You could try asking again and see what the current status is.

dpayet, 2017-07-30:
@casper They answer me few days ago :

> I checked with the Dev Team in this technical question. Under  Windows the driver can only be openCV, and it's slow. However under Linux you should be able to use v4l2loopback because we use API v4l2 to read the camera video stream.

> So to have a good performance we recommend using Linux with v4l2loopback instead of Windows with openCV, but we cannot guaranty that because have have never conducted any tests on that kind of user case.

So, according to them, it is possible to use v4l2loopback. But they don't explain how :)

The device created by v4l2loopback has a different major number than a real webcam. Perhaps a start ?

umlaeute, 2017-07-30:
Quoting dpayet <notifications@github.com>:
> The device created by v4l2loopback has a different major number than  
> a real webcam. Perhaps a start ?

no.
all drivers are assigned different major device numbers. this is  
expected behaviour.

afilini, 2017-08-08:
I'm working on this too, did someone manage to make some progress?

davidsierradz, 2017-09-30:
Same as https://github.com/umlaeute/v4l2loopback/issues/127#issuecomment-321054045?

umlaeute, 2017-10-01:
@davidsierradz ?

RodrigoOliveiraGoncalves, 2017-12-12:
?

casper, 2017-12-12:
No progress on my end. Have not tried Genymotion since last time I posted here. Keep sending them requests for this feature. Maybe they'll listen if enough people prod them.

yousifa, 2018-11-08:
@casper any progress on this?

casper, 2018-11-08:
Not using Genymotion at the moment, so I don't now what the status is. They have not contacted me either.

rdlmda, 2021-04-16:
Genymotion shows all my v4l2loopback devices listed on the camera selection screen, but it won't display the video stream, doesn't matter the video format I pick.