v4l2loopback/v4l2loopback #128: getting gst-launch-1.0 nvcamerasrc to work with v4l2loopback

kclim2, 2017-02-15:
* `v4l2loopback` version: v4l2loopback driver version 0.9.1

  * kernel version: Linux tegra-ubuntu 3.10.96-test #1 SMP PREEMPT Mon Dec 5 17:28:33 CST 2016 aarch64 aarch64 aarch64 GNU/Linux

  *lsb 
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04 LTS
Release:	16.04
Codename:	xenial

I made a post on nvidia developer forum on getting gst-launch-1.0 nvcamerasrc to work with v4l2loopback. Please help to shed some light if possible.

  https://devtalk.nvidia.com/default/topic/994012/jetson-tx1/passing-gst-launch-from-nvcamerasrc-to-v4l2sink/

umlaeute, 2017-02-15:
please try again with `v4l2loopback-0.10`

Apart from that, the `cannot identify device /dev/video3` error suggests that there is no `/dev/video3` file.
`v4l2loopback` will create device-files at the module-load stage (rather than dynamically).
So if you have a physical `/dev/video0`  and you want 3 loopback devices (`dev/video1`..`/dev/video3`) you ought to load the module like:

    sudo modprobe v4l2loopback devices=3

kclim2, 2017-02-15:
Hi umlaeute,

Thanks for the prompt response.

we originally using v4l2loopback-0.10 with nvidia TX1 Linux tegra-ubuntu 3.10.96 aarch64 GNU/Linux (see https://devtalk.nvidia.com/default/topic/988447/?comment=5081657 for the error detail).

we then realized v4l2loopback 0.9 doesn't give us the error (able to use guvcview to consume the created v4l2loopback adapter /dev/video3 with producer grabbing an USB webcam /dev/video0) but fail to get it working with nvcamerasrc at our /dev/video1 (see error as follow):

gst-launch-1.0 --gst-debug-level=3 nvcamerasrc ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)I420, framerate=(fraction)30/1' ! nvtee ! nvvidconv flip-method=2 ! 'video/x-raw(memory:NVMM), format=(string)I420, framerate=(fraction)30/1' ! v4l2sink device=/dev/video3
0:00:00.042475886 7498 0x59a0c0 ERROR GST_PIPELINE grammar.y:642:gst_parse_perform_link: could not link nvvconv0 to v4l2sink0

(see more error detail and nvcamerasrc' capability from https://devtalk.nvidia.com/default/topic/994012/?comment=5085953 )

PS: please ignore the error message produced by DaneLLL who was trying to help to resolve the issue (i believe DaneLLL didn't create his /dev/video3 at all)
> Apart from that, the cannot identify device /dev/video3 error suggests that there is no /dev/video3 file.
v4l2loopback will create device-files at the module-load stage (rather than dynamically).

Wish you could help to analyze the possible source of error.

umlaeute, 2017-02-15:
`v4l2loopback-0.10` fixed some major issues with `GStreamer-1.0`, I don't think that going back to `0.9` should be considered an option.

If `guvcview` doesn't work with `0.10` (but with `0.9`), then this should be fixed instead.

otoh, `guvcview` (2.0.4) works nicely here (my desktop machine, running Debian/sid on an amd64), and the all references to this program i found in the forums you linked to say:
>  guvcview has no problem to open the v4l2loopback device

so i'm not sure what the issue is that you are describing (apart from "it doesn't work")

kclim2, 2017-06-15:
the loopback device is created at /dev/video1

gst-launch gives the following errors:
gst-launch-1.0 v4l2src device=/dev/video1 ! xvimagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.265454948 14712       0x6fd940 ERROR          v4l2allocator gstv4l2allocator.c:1299:gst_v4l2_allocator_dqbuf:<v4l2src0:pool:src:allocator> buffer 1 was not queued, this indicate a driver bug.
0:00:00.265566042 14712       0x6fd940 WARN                 v4l2src gstv4l2src.c:884:gst_v4l2src_create:<v4l2src0> error: Failed to allocate a buffer
0:00:00.265689167 14712       0x6fd940 WARN                 basesrc gstbasesrc.c:2948:gst_base_src_loop:<v4l2src0> error: Internal data flow error.
0:00:00.265725156 14712       0x6fd940 WARN                 basesrc gstbasesrc.c:2948:gst_base_src_loop:<v4l2src0> error: streaming task paused, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate a buffer
Additional debug info:
gstv4l2src.c(884): gst_v4l2src_create (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0
Execution ended after 0:00:00.075286250
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

using guvcview --device=/dev/video1 is able to open the loopback device without giving any problem.

windzu, 2023-05-23:
> Hi umlaeute,
> 
> Thanks for the prompt response.
> 
> we originally using v4l2loopback-0.10 with nvidia TX1 Linux tegra-ubuntu 3.10.96 aarch64 GNU/Linux (see https://devtalk.nvidia.com/default/topic/988447/?comment=5081657 for the error detail).
> 
> we then realized v4l2loopback 0.9 doesn't give us the error (able to use guvcview to consume the created v4l2loopback adapter /dev/video3 with producer grabbing an USB webcam /dev/video0) but fail to get it working with nvcamerasrc at our /dev/video1 (see error as follow):
> 
> ```shell
> gst-launch-1.0 --gst-debug-level=3 nvcamerasrc ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)I420, framerate=(fraction)30/1' ! nvtee ! nvvidconv flip-method=2 ! 'video/x-raw(memory:NVMM), format=(string)I420, framerate=(fraction)30/1' ! v4l2sink device=/dev/video3
> 0:00:00.042475886 7498 0x59a0c0 ERROR GST_PIPELINE grammar.y:642:gst_parse_perform_link: could not link nvvconv0 to v4l2sink0
> ```
> 
> (see more error detail and nvcamerasrc' capability from https://devtalk.nvidia.com/default/topic/994012/?comment=5085953 )
> 
> Wish you could help to analyze the possible source of error.
> 
> PS: please ignore the error message produced by DaneLLL who was trying to help to resolve the issue (i believe DaneLLL didn't create his /dev/video3 at all)
> 
> > Apart from that, the cannot identify device /dev/video3 error suggests that there is no /dev/video3 file.
> > v4l2loopback will create device-files at the module-load stage (rather than dynamically).

Hello, I ran the same test on xavier and encountered the same problem as you, did you solve this problem in the end?

umlaeute, 2023-05-23:
liberal adding of `tee` elements often helps...

umlaeute, 2023-05-26:
i did a quick google-check and found [this](https://docs.nvidia.com/jetson/archives/r35.3.1/DeveloperGuide/text/SD/Multimedia/AcceleratedGstreamer.html?highlight=nvv4l2camerasrc#progressive-capture-using-nvv4l2camerasrc):
> The <kbd>nvv4l2camerasrc</kbd> plugin default currently supports only DMABUF (importer role) streaming I/O mode with `V4L2_MEMORY_DMABUF`.

the `v4l2loopback` device currently only supports `V4L2_MEMORY_MMAP`, so that might explain why it's not working.

i'm therefore closing this as a duplicate of #501

closed 2023-05-26 by umlaeute