v4l2loopback/v4l2loopback #129: error: no include path in which to search for stdint.h

Witiko, 2017-03-12:
### Step 1: Read this
snipped.

### Step 2: Describe your environment
  * `v4l2loopback` version: commit bfd990d5519710f57b6ca913ba6cac8b4e649918 (version 0.10.0)
  * kernel version: Linux inspiron 4.9.13 #1 SMP PREEMPT Sat Mar 11 11:23:48 CET 2017 x86_64 
  * Distribution (+version):
```
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 9.0 (stretch)
Release:        9.0
Codename:       stretch
```
  
### Step 3: Describe the problem

#### Steps to reproduce:

  1. `git clone 'https://github.com/umlaeute/v4l2loopback'`
  2. `cd v4l2loopback/`
  3. `make`
  
#### Observed Results:

```
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/witiko/tmp/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.13'
  CC [M]  /home/witiko/tmp/v4l2loopback/v4l2loopback.o
In file included from ./include/linux/media.h:26:0,
                 from ./include/media/media-entity.h:30,
                 from ./include/media/v4l2-dev.h:19,
                 from ./include/media/v4l2-common.h:29,
                 from /home/witiko/tmp/v4l2loopback/v4l2loopback.c:25:
/usr/lib/gcc/x86_64-linux-gnu/6/include/stdint.h:9:26: error: no include path in which to search for stdint.h
 # include_next <stdint.h>
                          ^
scripts/Makefile.build:304: recipe for target '/home/witiko/tmp/v4l2loopback/v4l2loopback.o' failed
make[2]: *** [/home/witiko/tmp/v4l2loopback/v4l2loopback.o] Error 1
Makefile:1507: recipe for target '_module_/home/witiko/tmp/v4l2loopback' failed
make[1]: *** [_module_/home/witiko/tmp/v4l2loopback] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.13'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
```
  
#### Expected Results:

I expected to successfully compile the kernel module.
  
#### Remarks:

The following is the contents of `/usr/lib/gcc/x86_64-linux-gnu/6/include/stdint.h`: 

```
#ifndef _GCC_WRAP_STDINT_H
#if __STDC_HOSTED__
# if defined __cplusplus && __cplusplus >= 201103L
#  undef __STDC_LIMIT_MACROS
#  define __STDC_LIMIT_MACROS
#  undef __STDC_CONSTANT_MACROS
#  define __STDC_CONSTANT_MACROS
# endif
# include_next <stdint.h>
#else
# include "stdint-gcc.h"
#endif
#define _GCC_WRAP_STDINT_H
#endif
```

umlaeute, 2017-03-12:
`stdint.h` is a standard header (C99)
if your system cannot use it, then your build-system is broken.
i don't think that there is anything that can be done on the v4l2loopback side.

i don't know why you are getting this error, but here's some thoughts:
- do you perchance use a non-standard compiler?
- do you perchance try to cross-compile the module for a different architecture?

closed 2017-03-12 by umlaeute

Witiko, 2017-03-12:
Indeed, I managed to mess up my headers. Sorry about wasting your time.

umlaeute, 2017-03-12:
thanks for confirming :-)

Witiko, 2017-03-12:
I also appologize to the readers with the same issue for not explaining how precisely I messed up my headers. The reason is that I do not know; it has something to do with mistakenly downgrading the `linux-headers` debian package and then manually updating it without triggering the update of the installed dkms packages. Rather than trace the exact cause, I reinstalled my custom headers and reran the triggers for the installed dkms packages manually:

```
# dpkg -i linux-headers-4.9.13_1.0.custom_amd64.deb
# apt install --reinstall `LC_ALL=C apt list 2>/dev/null | grep '\[installed[],]' | sed -n '/-dkms\//s#/.*##p'`
```