v4l2loopback/v4l2loopback #130: Error writing h.264 to loopback from ffmpeg

roger-, 2017-03-17:
I'm trying to transcode video from YUV to h.264 using ffmpeg. Just compiled the latest version of v4l2loopback and running `ffmpeg -f v4l2  -s 640x480 -an -r 5 -i /dev/video0 -c:v h264_omx -f v4l2 -an /dev/video1` on a Raspberry Pi  gives this:

```
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 164211.484936, bitrate: 24576 kb/s
    Stream #0:0: Video: rawvideo (YUY2 / 0x32595559), yuyv422, 640x480, 24576 kb/s, 5 fps, 5 tbr, 1000k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_omx))
Press [q] to stop, [?] for help
[h264_omx @ 0x2d483b0] Using OMX.broadcom.video_encode
[v4l2 @ 0x2d47060] V4L2 output device supports only a single raw video stream
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- ?z�p�?
Conversion failed!
```

If I remove the `-c:v h264_omx` part I don't get an error.

Any suggestions? Thanks!

umlaeute, 2017-03-18:
suggestion #1: please provide the information asked for in the issue-template, esp. v4l2looback-version, kernel-version and distribution (+version)

roger-, 2017-03-18:
Sure, here you go: Linux raspberrypi 4.4.50+, Raspbian Jessie and the current master of v4l2 loopback. 

Thanks!

PerryVanKuppeveld, 2017-06-14:
As far as I know you can't write h264 to v4l2 with ffmpeg, only raw. See the message V4L2 output device supports only a single raw video str

roger-, 2017-11-04:
Turns out it's pretty straightforward to modify ffmpeg to support writing H.264 video to V4L2 devices, not sure why this isn't done by default.

[Here](https://gist.github.com/roger-/def1a1df8b7bcc93baf67436f9f1aa58/revisions) are the changes that need to be made to `libavdevice/v4l2enc.c`.

closed 2017-11-04 by roger-

chikamichi, 2018-02-04:
Hi, would you make a PR out of this patch?

roger-, 2018-02-04:
Not sure if my patch is the best way to do things.

Unfortunately I’m a little busy at the moment and their patch process isn’t as simple as sending a PR.

umlaeute, 2018-02-05:
it might be best to open a ticket at the `ffmpeg` project, rather than discuss this here...

Manouchehri, 2020-05-09:
@roger- Thanks for the patch. I redid a bit of it and submitted it upstream to ffmpeg, it's merged now. https://github.com/FFmpeg/FFmpeg/commit/d0705ecb2bca2a72df13435251ce05c08d8142cc

roger-, 2020-05-09:
Awesome, thanks!