v4l2loopback/v4l2loopback #131: Cannot compile on debian 6 (squeeze)

mguijarr, 2017-03-31:
### Environment

Debian 6 'squeeze'

    uname -a
    Linux aelita 2.6.32-5-amd64 #1 SMP Sun May 6 04:00:17 UTC 2012 x86_64 GNU/Linux

    aelita:~/local/v4l2loopback % lsb_release -a
    No LSB modules are available.
    Distributor ID:	Debian
    Description:	Debian GNU/Linux 6.0.5 (squeeze)
    Release:	6.0.5
    Codename:	squeeze

### Step 3: Describe the problem:

When trying to make the project (`make`), I get the following error:

```
aelita:~/local/v4l2loopback` % make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/users/blissadm/local/v4l2loopback modules
make[1]: Entering directory `/usr/src/linux-headers-2.6.32-5-amd64'
  CC [M]  /users/blissadm/local/v4l2loopback/v4l2loopback.o
/users/blissadm/local/v4l2loopback/v4l2loopback.c:225: warning: ‘struct v4l2_ctrl’ declared inside parameter list
/users/blissadm/local/v4l2loopback/v4l2loopback.c:225: warning: its scope is only this definition or declaration, which is probably not what you want
/users/blissadm/local/v4l2loopback/v4l2loopback.c:226: error: variable ‘v4l2loopback_ctrl_ops’ has initializer but incomplete type
/users/blissadm/local/v4l2loopback/v4l2loopback.c:227: error: unknown field ‘s_ctrl’ specified in initializer
/users/blissadm/local/v4l2loopback/v4l2loopback.c:227: warning: excess elements in struct initializer
/users/blissadm/local/v4l2loopback/v4l2loopback.c:227: warning: (near initialization for ‘v4l2loopback_ctrl_ops’)
/users/blissadm/local/v4l2loopback/v4l2loopback.c:230: warning: initialization discards qualifiers from pointer target type
/users/blissadm/local/v4l2loopback/v4l2loopback.c:240: warning: initialization discards qualifiers from pointer target type
/users/blissadm/local/v4l2loopback/v4l2loopback.c:250: warning: initialization discards qualifiers from pointer target type
/users/blissadm/local/v4l2loopback/v4l2loopback.c:260: warning: initialization discards qualifiers from pointer target type
/users/blissadm/local/v4l2loopback/v4l2loopback.c:1227: warning: ‘struct v4l2_ctrl’ declared inside parameter list
/users/blissadm/local/v4l2loopback/v4l2loopback.c:1227: error: conflicting types for ‘v4l2loopback_s_ctrl’
/users/blissadm/local/v4l2loopback/v4l2loopback.c:225: error: previous declaration of ‘v4l2loopback_s_ctrl’ was here
/users/blissadm/local/v4l2loopback/v4l2loopback.c: In function ‘v4l2loopback_s_ctrl’:
/users/blissadm/local/v4l2loopback/v4l2loopback.c:1229: error: dereferencing pointer to incomplete type
/users/blissadm/local/v4l2loopback/v4l2loopback.c:1230: error: dereferencing pointer to incomplete type
/users/blissadm/local/v4l2loopback/v4l2loopback.c:1230: error: dereferencing pointer to incomplete type
/users/blissadm/local/v4l2loopback/v4l2loopback.c: In function ‘v4l2_loopback_init’:
/users/blissadm/local/v4l2loopback/v4l2loopback.c:2224: error: ‘struct v4l2_device’ has no member named ‘ctrl_handler’
make[4]: *** [/users/blissadm/local/v4l2loopback/v4l2loopback.o] Error 1
make[3]: *** [_module_/users/blissadm/local/v4l2loopback] Error 2
make[2]: *** [sub-make] Error 2
make[1]: *** [all] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-2.6.32-5-amd64'
make: *** [v4l2loopback.ko] Error 2
```

umlaeute, 2017-04-01:
hmm. as the README mentions, very old kernels are only supported by accident.

you might have luck using an older version of v4l2loopback, e.g. [0.7.1](https://github.com/umlaeute/v4l2loopback/releases/tag/v0.7.1) (which mentions an older 2.6.x kernel in the changelog, so it might just work with your specific kernel as well).

mguijarr, 2017-04-02:
Thanks a lot - I am going to try that :)

closed 2017-04-02 by umlaeute