v4l2loopback/v4l2loopback #132: unsupported format 'RGGB'

mguijarr, 2017-04-04:
* `v4l2loopback` version: 0.8.0
  * kernel version: 3.16
  * Distribution (+version): Debian 8 Jessie

I am now trying to do the same as described in this issue #32, except that I have BAYER RG 8
format (`fmt.pix.pixelformat = V4L2_PIX_FMT_SRGGB8`).

When I set the format, I can see the following:

Apr  4 10:49:59 lid30calva2 kernel: [326561.849989] v4l2-loopback[383]: unsupported format 'RGGB'
Apr  4 10:49:59 lid30calva2 kernel: [326561.849991] v4l2-loopback[902]: trying image 659x493
Apr  4 10:49:59 lid30calva2 kernel: [326561.849992] v4l2-loopback[942]: s_fmt_out(0) 0...1299548
Apr  4 10:49:59 lid30calva2 kernel: [326561.849993] v4l2-loopback[945]: outFOURCC=BGR4

So, it seems RGGB is not supported. As a consequence, when reading the video device in VLC for
example I do not get any valid image (just garbage).

Any idea ? Thanks in advance !

umlaeute, 2017-04-05:
yes `V4L2_PIX_FMT_SRGGB8` seems to be currently not supported.

you can easily add formats by editing the `v4l2loopback_formats.h` and filling in appropriate values.

i'll be happily accepting pull requests (once it works on your side).

mguijarr, 2017-04-05:
Ok, it sounds easy :) For the moment I am using RGB24, I will test with SRGGB8 as soon as I have time for this and I will submit a PR. Thanks a lot, and thanks for this excellent project!

closed 2018-05-28 by umlaeute