v4l2loopback/v4l2loopback #133: Cant insmod on lemaker guitar with kernel 3.10.100

iashwinprasada, 2017-04-08:
`Environment`

  * `v4l2loopback` version: latest 0.10

  * `kernel version:` Linux LeMaker 3.10.100 #1 SMP PREEMPT Sun Dec 18 06:36:37 PST 2016 armv7l armv7l armv7l GNU/Linux

    - Distributor ID:	Ubuntu
    - Description:	Ubuntu 16.04.1 LTS
    - Release:	16.04
  
`Problem:`

+ i'm trying to compile this driver on Lemaker guitar (http://www.lemaker.org/product-guitar-index.html)
+ i had to rollback to gcc 4.7 from gcc 5 because make wasnt working and used this link as reference
    - https://github.com/umlaeute/v4l2loopback/issues/108
+ the driver is getting compiled but i'm not able to insmod it, it throws an error 
    - insmod: ERROR: could not insert module v4l2loopback.ko: Invalid module format
+ i have installed the kernal headers for my distribution
   - http://wiki.lemaker.org/How_to_Compile_driver_on_Board

#### Observed Results:

make compiles fine and generates a .ko file but trying to insmod it doesnt work. i have tried insmod videodev before using v4l2src
  
What can i be doing wrong?

umlaeute, 2017-04-11:
you must:
- compile the module with the *same* kernel-headers as your target kernel
- compile the module with the same compiler as your target kernel
- compile for the correct target architecture (esp. do you cross-compile?)

iashwinprasada, 2017-04-11:
Sorry, i figured it out ! I had to use the right kernel headers, now it's working fine. Thank you.

closed 2017-04-11 by iashwinprasada