v4l2loopback/v4l2loopback #137: Gstreamer1.0 v4l2sink will not work at all for me.

redragonx, 2017-04-27:
### Step 2: Describe your environment

  * `v4l2loopback` version: 

```
[235036.841166] v4l2loopback driver version 0.10.0 loaded
[240131.013723] v4l2loopback driver version 0.10.0 loaded
```

  * kernel version: 

Linux X 4.10.10-1-ck-sandybridge #1 SMP PREEMPT Wed Apr 12 16:40:29 EDT 2017 x86_64 GNU/Linux

  * Distribution (+version):

```
LSB Version:    1.4
Distributor ID: Arch
Description:    Arch Linux
Release:        rolling
Codename:       n/a
```
  
### Step 3: Describe the problem:

Unable to use gstreamer1.0 to pass raw jpeg via http to v4l2sink device 0.

```
gst-launch-1.0 souphttpsrc location=http://192.168.1.125:80/live ! jpegdec ! videoconvert  ! v4l2sink device=/dev/video0    
Setting pipeline to PAUSED ...
libv4l2: error getting pixformat: Invalid argument
Pipeline is PREROLLING ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```

Also unable to use v4l2sink in its simple form.

```
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video0       
Setting pipeline to PAUSED ...
libv4l2: error getting pixformat: Invalid argument
Pipeline is PREROLLING ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```
In all cases I get `libv4l2: error getting pixformat: Invalid argument`

But other sinks will work with my source:

`gst-launch-1.0 souphttpsrc location=http://192.168.1.125:80/live ! jpegdec ! videoconvert ! xvimagesink`

```
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```

#### Steps to reproduce:

See above

#### Observed Results:

`libv4l2: error getting pixformat: Invalid argument`  

#### Expected Results:

Able to use v4l2sink

umlaeute, 2017-04-28:
what's the output of:

     v4l2-ctl --all  -d /dev/video1

also: as long as you have any consumer connected to the device, this will lock the format (size, ...); if your new producer requires a different format this will eventually fail.

redragonx, 2017-04-29:
```
                                                                                             ⍉
▶  v4l2-ctl --all  -d /dev/video0
Driver Info (not using libv4l2):
        Driver name   : v4l2 loopback
        Card type     : Dummy video device (0x0000)
        Bus info      : platform:v4l2loopback-000
        Driver version: 4.10.10
        Capabilities  : 0x85208003
                Video Capture
                Video Output
                Video Memory-to-Memory
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x85208003
                Video Capture
                Video Output
                Video Memory-to-Memory
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
Priority: 0
Video input : 0 (loopback: ok)
Video output: 0 (loopback in)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'YU12'
        Field             : None
        Bytes per Line    : 640
        Size Image        : 460800
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             : 
Format Video Output:
        Width/Height      : 640/480
        Pixel Format      : 'YU12'
        Field             : None
        Bytes per Line    : 640
        Size Image        : 460800
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             : 
Streaming Parameters Video Capture:
        Frames per second: 30.000 (30/1)
        Read buffers     : 8
Streaming Parameters Video Output:
        Frames per second: 30.000 (30/1)
        Write buffers    : 8

User Controls

                    keep_format (bool)   : default=0 value=0
              sustain_framerate (bool)   : default=0 value=0
                        timeout (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io (bool)   : default=0 value=0

```

After a few tries of trying new arguments, I realize that I can ignore the error. Though it is still annoying to see that error. :)

umlaeute, 2017-04-29:
ah yes; the point is that `libv4l2` tries to query the device for the current format.
if you do not have a producer attached to the loopback device, then the current format is undefined (because nobody defined it; and the loopback device is open to a wide variety of formats).

if you don't want to see the error, try to convince libv4l2 upstream to allow you to suppress it (or just patch it out :-))

closed 2017-04-29 by umlaeute

redragonx, 2017-04-29:
How would I specific a format here?

umlaeute, 2017-04-29:
by attaching a producer

now GStreamer uses libv4l2 for the `v4l2sink` element, which in turn will query the current format (before trying to set the requested format), which leads to the error.
not much you can do here.