v4l2loopback/v4l2loopback #139: SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175 on ubuntu16.04

ZhangWu1140, 2017-06-19:
I used this on ubuntu14.04 quite well for months and now i upgraded to 16.04 and meet this while make install:
```
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178

```
and here is something may helpful:

- `v4l2loopback` version:  0.10.0

- kernel version: Linux intel-desktop-developer123 4.4.0-79-generic #100-Ubuntu SMP Wed May 17 19:58:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

-  Distribution (+version): 
   No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial

__________________

as my research, this may caused by the kernel version 4.4 charging about vb2_queue, but i don't know where and how to fix this...

Thank you very much for any advise...

umlaeute, 2017-06-19:
is this actually a problem? e.g. does this prevent the module from being installed or inserted?

ZhangWu1140, 2017-06-20:
yes, i'm pretty sure about this, i used this `sudo modprobe -r v4l2loopback` to rmmod it and when i rebuilt it and here comes the peoblem.

umlaeute, 2017-06-20:
i still do not see any problem.
afaict, the error is unrelated and not harmful: during `make install` the system *also* tries to sign the module, and since you do not have the required keys to sign, this fails. nevertheless the `make install` succeeds, and you should be able the module.

if *this* error indeed prevents you from using the module, then the problem is with your kernel infrastructure, and must be fixed there (one could design a system that only allows modules signed by a given authority; in this case, not being able to create signed modules yourself is not a problem of the module code, but a social problem between you and that authority)

this is also a duplicate of https://github.com/umlaeute/v4l2loopback/issues/122.

closed 2017-06-20 by umlaeute

ZhangWu1140, 2017-06-21:
OK, i see now.  the word "error" in the build-logs just confused me, thank you very much!

ROBERT-MCDOWELL, 2018-11-10:
well, just encountered this issue on a unbuntu 32bit 14.05 in virtualbox
the module didn't install. any solution?

ROBERT-MCDOWELL, 2018-11-10:
I confirm that:
sudo cp -R  v4loopback /usr/src/v4l2loopback-1.1
sudo dkms add -m v4l2loopback -v 1.1
sudo dkms build -m v4l2loopback -v 1.1
sudo dkms install -m v4l2loopback -v 1.1
solve the issue