v4l2loopback/v4l2loopback #140: Doesn't start any more under Jessie 4.9 on a Raspberry Pi

fu-zhou, 2017-07-10:
v4l2loopback version: downloaded from GitHub today (July 10, 2017)
dmesg  | grep -i v4l2loopback doesn't return a version at it is not running

kernel version: 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

distribution:
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

Problem:
V4l2loopback driver doesn't get loaded during boot. It worked before upgrading to V4.9
The autostart is configured in /etc/modules-load.d/modules.conf:
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
v4l2loopback

That always worked with the previous version of Jessie

Log-File journalctl says:
Jul 10 23:36:27 ipcam systemd-modules-load[119]: Failed to find module 'v4l2loopback'

umlaeute, 2017-07-10:
you probably didn't compile the module and/or forgot to `make install`.

This is not a bug that can be fixed on the `v4l2loopback` side.

closed 2017-07-10 by umlaeute

fu-zhou, 2017-07-10:
Well, I did everything like in the previous version:

> root@ipcam:/v4l2loopback# make
> Building v4l2-loopback driver...
> make -C /lib/modules/`uname -r`/build M=/v4l2loopback modules
> make[1]: Entering directory '/usr/src/linux-headers-4.9.35-v7+'
>   Building modules, stage 2.
>   MODPOST 1 modules
> make[1]: Leaving directory '/usr/src/linux-headers-4.9.35-v7+'
> root@ipcam:/v4l2loopback# make install
> make -C /lib/modules/`uname -r`/build M=/v4l2loopback modules_install
> make[1]: Entering directory '/usr/src/linux-headers-4.9.35-v7+'
>   INSTALL /v4l2loopback/v4l2loopback.ko
>   DEPMOD  4.9.35-v7+
> make[1]: Leaving directory '/usr/src/linux-headers-4.9.35-v7+'
> root@ipcam:/v4l2loopback# modprobe v4l2loopback
> modprobe: FATAL: Module v4l2loopback not found.

Does it have anything to do ith the headers?

umlaeute, 2017-07-11:
you could try building with `make V=1` (also ´make install V=1`), to see what is really going un.

I suspect there might be a slight discrepancy between kernel headers and installed kernel.

in any case, I still think that this is a problem with raspbian rather than v42loopback.