v4l2loopback/v4l2loopback #146: V4L2_MEMORY_USERPTR Support

cnalley, 2017-09-04:
Are there any plans in the future to add support for streaming I/O with user pointers? I have a device that I need to mimic that uses user pointer I/O so I may try to work on this functionality in the future. Is there a reason user pointer support was not included and has there been any work or discussion towards adding this feature?

umlaeute, 2017-09-05:
patches welcome

Liza1444, 2023-08-22:
Hello,
as far as i see userptr not supported by v4l2loopback.
is there any planing on supporitng it ?

any idea what are the main chanllages supporting that feature or why is was not implemented so far ?
coluf anyone please advice a reference that would help with it ?

my main goal is to write app that works with v4l2 and user ptr .

movepointsolutions, 2023-08-29:
https://github.com/movepointsolutions/v4l2loopback
I'm starting doing userptr support,
examples and particular gst-launch-1.0 pipelines welcome

umlaeute, 2023-08-29:
looking forward to your PR