v4l2loopback/v4l2loopback #148: Modprobe v4l2loopback. Unknown symbol in module, or unknown parameter. Amazon EC2 instance.

maxfagotto, 2017-11-02:
### Step 2: Describe your environment
  * `v4l2loopback` version: _____
sudo dmesg  | grep -i v4l2loopback
[246726.856439] v4l2loopback: Unknown symbol video_devdata (err 0)
[246726.856446] v4l2loopback: Unknown symbol v4l2_ctrl_new_custom (err 0)
[246726.856452] v4l2loopback: Unknown symbol video_unregister_device (err 0)
[246726.856459] v4l2loopback: Unknown symbol video_device_alloc (err 0)
[246726.856467] v4l2loopback: Unknown symbol v4l2_device_register (err 0)
[246726.856480] v4l2loopback: Unknown symbol __video_register_device (err 0)
[246726.856485] v4l2loopback: Unknown symbol v4l2_ctrl_handler_free (err 0)
[246726.856489] v4l2loopback: Unknown symbol v4l2_device_unregister (err 0)
[246726.856493] v4l2loopback: Unknown symbol video_device_release (err 0)
[251354.079159] v4l2loopback: Unknown symbol video_ioctl2 (err 0)
[251354.079173] v4l2loopback: Unknown symbol v4l2_ctrl_handler_init_class (err 0)

 * kernel version: _____
uname -a
Linux ip-172-31-28-88 4.4.0-1038-aws #47-Ubuntu SMP Thu Sep 28 20:05:35 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

 * Distribution (+version): _____
 lsb_release -a
 No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

### Step 3: Describe the problem:

I installed the package with aptitude install v4l2loopback-dkms but after I tried to load the module with sudo modprobe v4l2loopback I got this error message:

modprobe: ERROR: could not insert 'v4l2loopback': Unknown symbol in module, or unknown parameter (see dmesg)

Thanks

umlaeute, 2017-11-05:
It seems that some dependency modules are missing (e.g. `videodev`).
Run `depmod -a` to refresh the dependency chain.

I wonder why this isn't done automatically by dkms:
- which version of `v4l2loopback` are you installing? (apt should tell)
- which version of `dkms` are you using? (apt should tell)

maxfagotto, 2017-11-05:
I ran depmod -a but nothing has changed. 

Below the answers to your questions:

#apt-show-versions v4l2loopback-dkms
v4l2loopback-dkms:all/xenial 0.9.1-4 uptodate

#apt-show-versions dkms
dkms:all/xenial-updates 2.2.0.3-2ubuntu11.5 uptodate

Thanks

umlaeute, 2017-11-05:
is the `videodev` module loaded? (if not, load it manually *before* loading `v4l2loopback`)

umlaeute, 2017-12-06:
assuming that manually loading the `videodev` module fixes the issue, i'm closing this.

closed 2017-12-06 by umlaeute

cetinakat, 2018-03-17:
I found a solution that if you run sudo apt-get install kmod module-init-tools, it will load videodev and after that you can run again modprobe v4l2loopback.

umlaeute, 2018-03-17:
you load it like any other module: `modprobe <modulename>`.