v4l2loopback/v4l2loopback #150: Compiled fine but not able to run on Raspbian Stretch (linux-headers-4.9.63-v7+)

pavangayakwad, 2017-11-21:
I am able to compile (git source downloaded) and install successfully. However, when I run modprobe get this error: FATAL: Module v4l2loopback not found in directory /lib/modules/4.9.63-v7+

[image]

`sudo apt get install v4l2loopback-dkmg` also ran fine and when I run modprobe get the same error mentioned above.

Any help would be greatly appreciated as I am stuck here now!

umlaeute, 2017-11-21:
1st: please always post text instead of screenshots!

then, re-run the installation in verbose mode:

     sudo make install V=1

and check whether there *is* a `v4l2loopback.ko` file in the installation directory (something like `/lib/modules/4.9.63-v7+/extra/`).

Check the type of this file.

    find /lib/modules/ -name v4l2loopback.ko -exec file {} +

Finally check whether loading the module directly succeeds:

    sudo modprobe /lib/modules/4.9.63-v7+/extra/v4l2loopback.ko

(adjust the paths as needed)

ashirvadz, 2018-02-25:
I've run into the same issue. 
Fixed by just running:

`sudo depmod -a`

closed 2018-02-26 by umlaeute