v4l2loopback/v4l2loopback #151: FFMPEG video0 (webcam) -> video2 (loopback) video3 (loopback)

jensJJ, 2017-11-30:
### Step 2: Describe your environment

  * `v4l2loopback` version:

[12420.914648] v4l2loopback: loading out-of-tree module taints kernel.
[12420.916962] v4l2loopback driver version 0.9.1 loaded

  * kernel version: 

Linux tinkerboard 4.13.16-rockchip #19 SMP PREEMPT Fri Nov 24 19:34:54 CET 2017 armv7l armv7l armv7l GNU/Linux

  * Distribution (+version): 

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

  
### Step 3: Describe the problem:

I am trying to stream the video0 (webcam USB) to 2x virtual loopback video2 and video3 using FFMPEG. I am trying to access video2 or video3 using OpenCV while streaming and it is not receiving any video data.

#### Steps to reproduce:

  1. v4l2-ctl --list-devices
   rockchip-rga ():
	/dev/video1

Dummy video device (0x0000) (platform:v4l2loopback-000):
	/dev/video2

Dummy video device (0x0001) (platform:v4l2loopback-001):
	/dev/video3

HD USB Camera (usb-ff540000.usb-1.1):
	/dev/video0

  2. ffmpeg -f v4l2 -i /dev/video0 -f v4l2 /dev/video2 -f v4l2 /dev/video3

  3. Python File OpenCV

  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

  ```
  // TODO(you): code here to reproduce the problem
  ```

umlaeute, 2017-12-01:
does it work if you only output to a single virtual device?

jensJJ, 2017-12-01:
Yes, I also tried with one only. Also following this example:

https://www.timdejong.nl/blog/use-webcam-two-applications-under-linux-simultaneously-using-v4l2loopback

It says that video0 is busy, so something must be run. But I cant get video2.

umlaeute, 2017-12-01:
if it does work with only one device (which you seem to confirm), then I think the bug is in `ffmpeg` (resp. `OpenVC`).
can you play back the loopback devices with `vlc`

for the timdejog example, this works for me. i'm not sure what you mean with 
> It says that video0 is busy, so something must be run
the example obviously requires that no other application access `/dev/video0` at the same time.

jensJJ, 2017-12-01:
I mean I have tried to loopback to `/dev/video2` only. And it does not work when I try to ?
`VideoCapture(2)` inside OpenCV. 

I just wanted to check if ffmpeg is doing something on `video0`.

Which example should I try with vlc?

jensJJ, 2017-12-01:
I tried `vlc v4l2:///dev/video0:width=800:height=600`  . VLC opens, but is not showing a picture. Then I tried `video2` with using dejong v4l2loopback, I get `not a videocapture device`.

jensJJ, 2017-12-11:
I made it work. I had to set to `-input_format mjpeg and -pix_fmt yuyv422`.

umlaeute, 2017-12-11:
cool. thanks for the heads up.

closed 2017-12-11 by umlaeute