v4l2loopback/v4l2loopback #152: Skype 8.11 does not work with v4l2loopback

albmi, 2017-12-04:
### Step 2: Describe your environment

  * `v4l2loopback` version: 

`[  150.188253] v4l2loopback: loading out-of-tree module taints kernel.
[  150.188306] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
[  150.189042] v4l2loopback driver version 0.10.0 loaded`

  * kernel version: 

`Linux ubuntu 4.10.0-40-generic #44~16.04.1-Ubuntu SMP Thu Nov 9 15:37:44 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

  * Distribution (+version): 

`No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial`
  
### Step 3: Describe the problem:

I have been using this module for a few months and never had a problem using it with Skype.
The version of Skype I had installed was 4.3.0.37.
However in the beginning of November, this version was discontinued and I was forced to update to version 8.11.0.4.
This new version does not seem to support virtual devices created with v4l2loopback module.
The virtual cameras are recongnized by the Skype application, but when trying to preview the video stream, the virtual camera feed is black.

#### Steps to reproduce:

  1. Insert module
`sudo insmod v4l2loopback.ko devices=1 exclusive_caps=1` or `sudo insmod v4l2loopback.ko`
  2. Redirect video from the physical device 0 to virtual device 1
`ffmpeg -f v4l2 -i /dev/video0 -f v4l2 /dev/video1`
Alternatively I could use a video file
`ffmpeg -f mp4 -re -i /video.mp4 -f v4l2 /dev/video1`
  3. Open skype
`skypeforlinux`
  
#### Observed Results:

 When trying to preview the virtual camera feed, the video is all black.
If I'm redirecting video from the physical camera, then the physical camera will also have the video feed all black, when previewing it on skype.
If a video file is used, then the physical camera works normally.

Image for reference.
[image]
  
#### Expected Results:

 I expected skypeforlinux to be able to read the video stream from the virtual device.

I understand this problem is probably related entirely with skype, however I was expecting to see if someone had encountered the same issue. And if so, how could I solve it.

I've tested several of the methods already used in other issues related to v4l2loopback repository. None seem to solve my problem.
I've also followed the wiki regarding skype, which also did not solve the problem.
I have also tried to use the newest ubuntu version, which did not help.

Thank you in advance.

umlaeute, 2017-12-05:
i'm not sure i understand these cases:

> - When trying to preview the virtual camera feed, the video is all black.
> - If I'm redirecting video from the physical camera, then the physical camera will also have the video feed all black, when previewing it on skype.
> - If a video file is used, then the physical camera works normally.

i assume that with "preview" you mean "preview within skype" (as in your screenshot).
i don't understand the second case (are you trying to access '/dev/video0` simulatenously by both `skype` and  `ffmpeg`? this is mostly likely not going to work)
does the third case mean, that you *can* use a v4l2loopback device with skype (e.g. when fed from a video-file), it only doesn't work when feeding it from `/dev/video0`?

albmi, 2017-12-05:
Yes when I mean preview, is preview within skype.
For the second case I was just providing an example, where the physical camera won't work, since it is being used by both ffmpeg and skype.
In the third case I was trying to explain that the physical camera does work if it is not being used by skype and ffmpeg simultaneously.
But the virtual device /dev/video1 never works.

Eoksni, 2017-12-05:
I can confirm - I'm having the same issue. I don't know which version of skype/v4l2loopback caused it to break - I never tried v4l2loopback before. 

With google hangouts its working fine.

Fedora 27, skype 8.11

angel200sdnot, 2017-12-06:
This is also happening to me, before updating skype it was working fine

umlaeute, 2017-12-06:
so to conclude: this appears to be a problem in skype.
please report bugs at their bugtracker.

closed 2017-12-06 by umlaeute