v4l2loopback/v4l2loopback #154: Kernel not found

tomforsyth90, 2018-01-18:
4.9.76-v7+ this is my linux kernel version and when I tried to install using sudo aptitude install v4l2loopback-source module-assistant and sudo module-assistant auto-install v4l2loopback-source it said a version for my kernel could not be found... am I doing this wrong? Thanks

umlaeute, 2018-01-19:
this is not an issue of `v4l2loopback`: as the README explains you must have the kernel headers for your target kernel (usually the one your are running) installed.

apart from that, on deb-based systems the preferred way is to install the `dkms` module.

closed 2018-01-19 by umlaeute

tomforsyth90, 2018-01-19:
I appreciate the reply, could you please explain to me what I am required to do to ensure the correct kernel headers are installed and if not how I can make sure the right ones are installed as I have just installed the module via the dkms way that you stated above and this is what it said:
DKMS: install completed.
Module build for kernel 4.9.59+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 4.9.59-v7+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 4.9.76+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 4.9.76-v7+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Setting up linux-headers-3.6-trunk-rpi (3.6.9-1~experimental.1+rpi7) ...
Examining /etc/kernel/header_postinst.d.
run-parts: executing /etc/kernel/header_postinst.d/dkms 3.6-trunk-rpi

Is this installed correctly or not? Any help would be very appreciated as not all of us are as knowledgeable as yourself in regards to this.  Thanks