v4l2loopback/v4l2loopback #157: v4l2sink element always failed with "Cannot identify device '/dev/video'"

golgobot, 2018-03-02:
### Environment

  * `v4l2loopback` version: 0.10.0
  * kernel version: Linux Air-Math-Melon-Towel 3.10.104 #1 SMP PREEMPT Thu Mar 1 19:02:35 EST 2018 armv7l GNU/Linux

### The Problem:

Built v4l2loopback in buildroot targeting NVidia TK1 (Jetson Board). Loaded the kernel module. But when I use the v4l2sink element in a gstreamer pipeline, I always get the same error: `ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Cannot identify device '/dev/video0'.` But the video devices definitely exist after loading the module

```bash
# ls -l /dev/video*
crw-rw----    1 root     video      81,   0 Mar  2 15:55 /dev/video0
crw-rw----    1 root     video      81,   1 Mar  2 15:55 /dev/video1
crw-rw----    1 root     video      81,   2 Mar  2 15:57 /dev/video2
```

#### Steps to reproduce:

  1. `modprob v4l2loopback`
  2. `gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video0`
  
#### Observed Results:
```bash
# gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video0
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Cannot identify device '/dev/video0'.
Additional debug info:
v4l2_calls.c(606): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
system error: No such file or directory
Setting pipeline to NULL ...
Freeing pipeline ...
```
  
#### Expected Results:

  * Expected a test video to be piped at `dev/video0` so a local web browser can open that device and display the test video.

golgobot, 2018-03-02:
Was trying to stream to a device that was already open

closed 2018-03-02 by golgobot

mcan06, 2021-12-01:
How can u solve this problem, I got the same error when I'm using with GNURadio?

umlaeute, 2021-12-01:
most likely by picking a device that was not already used...

aiui, the original problem was just a confusion which device was actually the loopback device, and once figuring that out, everything worked.

mcan06, 2021-12-01:
Actually, I couldn't understand what do you mean, can you extend what did you mention?

umlaeute, 2021-12-01:
the discussion ended with:
> Was trying to stream to a device that was already open

here's a couple of explanations for this sentence:

- OP was trying to stream to a given device (e.g. `/dev/video0`)
- the device was already in use (by some other process)
- stopping the other process would free the device
- the device then became usable

OR

- OP was trying to stream to a given device (e.g. `/dev/video0`)
- the device was already in use (by some other process)
- OP then made sure the other process wasn't started in the first place
- the device then became usable

OR

- OP was trying to stream to a given device (e.g. `/dev/video0`)
- the device was not a loopback device in the first place
- OP found out which device was the loopback device (e.g. `/dev/video2`)
- using the actual device was successful