v4l2loopback/v4l2loopback #158: new release? (because timer api change)

xlotlu, 2018-03-06:
Hi guys, this isn't really a bug report because you fixed things already:
4.15 is starting to land in distros, and building fails because of the api change which you handled in https://github.com/umlaeute/v4l2loopback/commit/00eb0df20248d7a37594905d64030d0bd0b72eb8.

I'm gonna file a bug report with debian about this. Do you intend to make a release soon, or should I ask them to apply the patch?

P.S.: thanks for the software.

  * `v4l2loopback` version: 0.10.0
  * kernel version: 4.15.4
  * Distribution (+version): Debian unstable

```
Building v4l2-loopback driver...
make -C /lib/modules/4.15.0-1-amd64/build M=/var/lib/dkms/v4l2loopback/0.10.0/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-1-amd64'
  CC [M]  /var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.c: In function ‘v4l2_loopback_init’:
/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.c:2206:2: error: implicit declaration of function ‘setup_timer’; did you mean ‘setup_irq’? [-Werror=implicit-function-declaration]
  setup_timer(&dev->sustain_timer, sustain_timer_clb, nr);
  ^~~~~~~~~~~
```

closed 2018-03-20 by umlaeute

shark0der, 2018-05-29:
Are there any news regarding this? Almost 3 months have passed and I'm still unable to build it. I'm using Ubuntu 16.04 with 4.15.0-22-generic.

umlaeute, 2018-05-29:
this has been *closed* because the issue was supposedly fixed with the `v0.11.0` release, 2 months ago.

shark0der, 2018-05-29:
I was able to build by cloning the repo. The xenial release is outdated: `0.9.1-4`.

shark0der, 2018-05-31:
By default, it indeed comes with `4.4`. Not entirely sure what you mean by **manually**. I did not compile or manually download a new kernel - I've installed the `linux-image-generic-hwe-16.04-edge` package which comes with `4.15` (it's in the official repo).

umlaeute, 2018-06-01:
well in any case, your problem is (was) an ubuntu packaging issue. unrelated to `v4l2loopback` upstream development (or release cycles).

shark0der, 2018-06-01:
Understood. Thanks.