v4l2loopback/v4l2loopback #159: make fails - no such file or directory

jgiesen, 2018-03-06:
Environment:

  * `v4l2loopback` version: _____

          "sudo dmesg  | grep -i v4l2loopback" - gives no output

  * kernel version: Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

  * Distribution (+version): Raspbian GNU/Linux 9.3 (stretch)

Problem:
I am new to linux and tried to follow the install instructions. But i get this message, when executing "make && sudo make install" :

> Building v4l2-loopback driver...
> make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback modules
> make[1]: *** /lib/modules/4.9.59-v7+/build: No such file or directory.  Stop.
> Makefile:42: recipe for target 'v4l2loopback.ko' failed

There really is no folder build in "/lib/modules/4.9.59-v7+"
I guess it is the installed kernel headers version don't match. But i don't know how to correct this.

Could someone be so kind and help me?

Thanks alot!

jgiesen, 2018-03-06:
"sudo apt-get install raspberrypi-kernel-headers" installed the missing headers.
but now, running "modprobe v4l2loopback",  i get the error:

modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.9.59-v7+

umlaeute, 2018-03-06:
you might want to follow the [README](https://github.com/umlaeute/v4l2loopback/blob/master/README.md#install)

closed 2018-03-06 by umlaeute