v4l2loopback/v4l2loopback #161: Failed to allocate a buffer when using as part of videomixer

depauwjimmy, 2018-03-09:
### Environment

  * `v4l2loopback` version: 0.10.0 (compiled from master as of March 9)
  * GStreamer version: 1.8.3
  * kernel version: Linux tegra-ubuntu 4.4.38-tegra aarch64
  * Distribution (+version): Ubuntu 16.04.4 LTS
  
### Describe the problem:

I have been trying to use a Camera as a video live source in a gstreamer Pipeline.
It works perfectly when using as standalone video source but as soon as i want to combine it with another video using the videomixer plugin, it does not work.
It actually does not matter that the source are real camera or not, using videotestsrc for both have the same result.

#### Steps to reproduce:

  1. Feed the device using videotestsrc

`gst-launch-1.0 videotestsrc pattern=ball ! "video/x-raw, format=(string)I420, width=(int)1024, height=(int)768" ! v4l2sink device=/dev/video9`

  2. Run the videomixer pipeline

`gst-launch-1.0 videomixer sink_1::xpos=100 sink_1::ypos=100 name=mixer ! fakesink v4l2src name=LoopBack device=/dev/video9 ! "video/x-raw, width=(int)1024, height=(int)768, format=(string)I420" ! queue ! mixer.sink_0 videotestsrc pattern=snow ! "video/x-raw, width=(int)640, height=(int)480, format=(string)I420, framerate=(fraction)30/1" ! queue ! mixer.sink_1`
  
#### Observed Results:

The pipeline fails to start with the following logs

```
0:00:00.143583751  2676       0x622370 ERROR          v4l2allocator gstv4l2allocator.c:1299:gst_v4l2_allocator_dqbuf:<LoopBack:pool:src:allocator> buffer 0 was not queued, this indicate a driver bug.
0:00:00.143640135  2676       0x622370 WARN                 v4l2src gstv4l2src.c:884:gst_v4l2src_create:<LoopBack> error: Failed to allocate a buffer
0:00:00.143685031  2676       0x622370 INFO        GST_ERROR_SYSTEM gstelement.c:1879:gst_element_message_full:<LoopBack> posting message: Failed to allocate a buffer
0:00:00.143761894  2676       0x622370 INFO        GST_ERROR_SYSTEM gstelement.c:1902:gst_element_message_full:<LoopBack> posted error message: Failed to allocate a buffer
0:00:00.143787142  2676       0x622370 INFO                 basesrc gstbasesrc.c:2739:gst_base_src_loop:<LoopBack> pausing after gst_base_src_get_range() = error
0:00:00.143917926  2676       0x622370 WARN                 basesrc gstbasesrc.c:2948:gst_base_src_loop:<LoopBack> error: Internal data flow error.
0:00:00.143933861  2676       0x622370 WARN                 basesrc gstbasesrc.c:2948:gst_base_src_loop:<LoopBack> error: streaming task paused, reason error (-5)
0:00:00.143958885  2676       0x622370 INFO        GST_ERROR_SYSTEM gstelement.c:1879:gst_element_message_full:<LoopBack> posting message: Internal data flow error.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:LoopBack: Failed to allocate a buffer
0:00:00.144011109  2676       0x622370 INFO        GST_ERROR_SYSTEM gstelement.c:1902:gst_element_message_full:<LoopBack> posted error message: Internal data flow error.
```
  
#### Expected Results:

If i don't use a source from a v4l2loopback device it works as expected. For instance the following pipeline

```
gst-launch-1.0 videomixer sink_1::xpos=100 sink_1::ypos=100 name=mixer ! videoconvert ! "video/x-raw, width=(int)1024, height=(int)768, format=(string)I420" ! nvjpegenc quality=100 ! rtpjpegpay ! udpsink host=192.168.0.20 port=1234 videotestsrc pattern=snow ! "video/x-raw, width=(int)1024, height=(int)768, format=(string)I420" ! queue ! mixer.sink_0 videotestsrc pattern=ball ! "video/x-raw, width=(int)640, height=(int)480, format=(string)I420, framerate=(fraction)30/1" ! queue ! mixer.sink_1
```

umlaeute, 2018-03-09:
most likely this is a duplicate of #83 (if not, please complain)

closed 2018-03-09 by umlaeute

stephematician, 2025-01-31:
I don't know if this is a duplicate of #83. This still fails with the updated __compositor__ pad, e.g.

```console
user@home:~$ sudo modprobe v4l2loopback max_buffers=4 video_nr=10
user@home:~$ gst-launch-1.0 videotestsrc pattern=ball ! \
	"video/x-raw, format=(string)I420, width=(int)1024, height=(int)768" ! \
	v4l2sink device=/dev/video10
user@home:~$ gst-launch-1.0  v4l2src device=/dev/video10 ! \
	"video/x-raw, width=(int)1024, height=(int)768, format=(string)I420" ! \
	compositor name=comp sink_0::alpha=0.7 sink_1::alpha=0.5 ! \
	videoconvert ! xvimagesink \
	videotestsrc ! \
	video/x-raw,format=AYUV,framerate=\(fraction\)5/1,width=320,height=240 ! comp.
```