v4l2loopback/v4l2loopback #162: Can't make v4l2loopback

dimitri2705, 2018-03-10:
Hello,

I'm trying since the start of the day to figure out how to make this work. 
When I try to "make" the sources, an error occur.

```
root@orangepilite:~/Desktop/camera/v4l2loopback# make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/root/Desktop/camera/v4l2loopback modules
make[1]: *** /lib/modules/3.4.113-sun8i/build: No such file or directory.  Stop.
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2

```

I know that the issue aready exist, but the solutions doesn't helped me.
Any help is welcome

Thanks !

umlaeute, 2018-03-10:
you obviously don't have the kernel-headers installed for your specific kernel.

closed 2018-03-10 by umlaeute

LegendHector, 2020-04-22:
hi there how do i get the kernal header installed

ram-sh, 2020-05-12:
> 
> 
> hi there how do i get the kernal header installed

+1

umlaeute, 2020-05-12:
ask you distributor ;-)

seriously. the exact way differs from distro to distro; there's no generic way to explain it.

(and +1 posts should just use the emoji...)