v4l2loopback/v4l2loopback #164: Some question

lddd, 2018-03-14:
Hey there,

are there any specific requirements if would need to take care of for having v4l2loopback simulating a machines desktop/screen as the camera input? 

I am looking for a solution where i am able to emulate a android device for e.g by using something like Genymotion (supporting Camera).

And using v4l2loopback and ffmpeg for e.g to replace the video output with the servers desktop instead.

Now can anyone tell me if this is possible with any Amazon AWS EC instance or do i need to rent some other specific dedicated server for it? 

thanks in advance,
Roman

umlaeute, 2018-03-14:
sure, it's possible to setup a video-provider that captures the desktop and feeds into a `v4l2loopback` device.
this requires (of course) to have a *desktop* running on your system, on top of some "painting infrastructure" like `X11`, `X.org` or `VNC`.

whether this can be done on AWS EC, is a question that you need to ask the AWS EC support.

closed 2018-03-14 by umlaeute

lddd, 2018-03-14:
Thanks i am currently testing it on a root server of mine. Will let you know.