v4l2loopback/v4l2loopback #165: v4l2loopback with emulator?

lddd, 2018-03-17:
Hi there,

i am trying to find a solution to replace the camera input for some emulated device (android studio emulator) with the machines desktop screen or any static image.

Do you know if this is possible. I tried this on my Mac with using something called "CamTwist" which also able to broadcast my desktop as video source.

However inside the Emulator the Camera still shows the real webcam output. 

Any idea?

umlaeute, 2018-03-17:
i don't know whether it is possible, but in any case, it is an issue of the emulator software itself (that is: whether the emulator allows you to select a videodevice to "pass thru" to the emulated machine).

i've successfully used v4l2loopback as a source for a webcam inside VirtualBox machines.
i don't know about android studio emulator.

(i don't know what your Mac has to do with it.
the emulator needs to access the virtual video device on your host system. since `v4l2loopback` only works on Linux, this basically means that your host system must be Linux if you want to use it that way.)

closed 2018-03-17 by umlaeute

lddd, 2018-03-18:
And v4l2loopback with the android studio emulator?

juanpmarin, 2018-08-08:
@umlaeute What do you did to use the camera in a virtualbox's vm ?

umlaeute, 2018-09-07:
@juanpmarin i re-checked and I cannot find it any more. maybe i was mistaken, and i just passed through my USB-device.