v4l2loopback/v4l2loopback #166: ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate a buffer

TheSalarKhan, 2018-03-19:
### Environment

  * `v4l2loopback` version: 0.10.0

  * kernel version: Linux XXXX 4.13.0-37-generic #42-Ubuntu SMP Wed Mar 7 14:13:23 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

  * Distribution (+version): 
```
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Pop!_OS 17.10 (Artful Aardvark)
Release:        17.10
Codename:       artful
```
  * Relevant Environment variables
```
export GST_DEBUG="*:2"
```

  * Installation
```
sudo apt-get install v4l2loopback-dkms
```

### Problem:
Can't get local camera /dev/video0 to be copied to loopback device /dev/video1.

#### Steps to reproduce:
1) Load module
```
sudo modprobe v4l2loopback devices=2 exclusive_caps=0 max_buffers=8
```
2) Start the producer
```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! v4l2sink device=/dev/video1
```
3) Start the consumer
```
gst-launch-1.0 -v v4l2src device=/dev/video1 ! xvimagesink
```

#### Observed Results:

  * PRODUCER: Although with a few warnings the producer starts fine, and the camera also lights up to indicate capture.
```
Setting pipeline to PAUSED ...
libv4l2: error getting pixformat: Invalid argument
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:02.953041493 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at YUYV@48x32: 0
0:00:02.953148045 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at UYVY@48x32: 0
0:00:02.953222466 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at 422P@48x32: 0
0:00:02.953294719 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at YU12@48x32: 0
0:00:02.953379735 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at YV12@48x32: 0
0:00:02.953452392 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at 411P@48x32: 0
0:00:02.953522466 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at YVU9@48x32: 0
0:00:02.953607133 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at YUV9@48x32: 0
0:00:02.953685179 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at JPEG@48x32: 0
0:00:02.953751668 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at MJPG@48x32: 0
0:00:02.953819413 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at dvsd@48x32: 0
0:00:02.953909041 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at RGB4@48x32: 0
0:00:02.953971961 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at BGR4@48x32: 0
0:00:02.954037298 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at RGB3@48x32: 0
0:00:02.954135606 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at BGR3@48x32: 0
0:00:02.954234993 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at RGBQ@48x32: 0
0:00:02.954452737 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at RGBP@48x32: 0
0:00:02.954563363 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at RGBO@48x32: 0
0:00:02.954668238 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at GREY@48x32: 0
0:00:02.954781713 10416 0x55fa53353400 WARN                    v4l2 gstv4l2object.c:2492:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0> Unknown frame interval type at VP80@48x32: 0
[... 19 lines cut ...]
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-raw, format=(string)YV12, width=(int)1280, height=(int)720, framerate=(fraction)100/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1
/GstPipeline:pipeline0/GstXvImageSink:xvimagesink0.GstPad:sink: caps = video/x-raw, format=(string)YV12, width=(int)1280, height=(int)720, framerate=(fraction)100/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1
0:00:00.066949960 10441 0x55d87f516940 ERROR          v4l2allocator gstv4l2allocator.c:1304:gst_v4l2_allocator_dqbuf:<v4l2src0:pool:src:allocator> buffer 1 was not queued, this indicate a driver bug.
0:00:00.067014285 10441 0x55d87f516940 WARN                 v4l2src gstv4l2src.c:866:gst_v4l2src_create:<v4l2src0> error: Failed to allocate a buffer
0:00:00.067139322 10441 0x55d87f516940 WARN                 basesrc gstbasesrc.c:2939:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:00:00.067175779 10441 0x55d87f516940 WARN                 basesrc gstbasesrc.c:2939:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate a buffer
Additional debug info:
gstv4l2src.c(866): gst_v4l2src_create (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0
Execution ended after 0:00:00.037965179
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

  * NOTE: The consumer works fine if we replace the producer with:
```
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video1
```

#### Expected Results:

  * The consumer should start and display a video on a window.

TheSalarKhan, 2018-03-19:
Update.... So I tried using a website for an online webcam test: https://www.onlinemictest.com/webcam-test/. and the device there seems to be working fine. I get my webcam view, and the video goes away when I kill the producer instance.

homespring, 2018-11-22:
I have exactly the same issue in my environment:
- Ubuntu Mate 18.04, kernel version 4.15.0-39
- Gstreamer 1.14.1
- v4l2loopback version 0.10.0

Producer pipeline (starts and works fine):
`gst-launch-1.0 -vv videotestsrc ! video/x-raw,width=640,height=480,format=I420 ! identity drop-allocation=1 ! v4l2sink device=/dev/video100`

Consumer pipeline:
`gst-launch-1.0 -vv v4l2src device=/dev/video100 ! video/x-raw,width=640,height=480,format=I420 ! autovideosink sync=false`

This pipeline shows only one frame and crashes with error:
> ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate a buffer

umlaeute, 2018-11-23:
@homespring so please try `v4l2loopback-0.12.0`

nasafix-nasser, 2019-02-19:
Same thing with version 0.12.1 on my ARM based embedded platform.
Note that if I change the producer to use a videotestsrc, there is no problem at all.

The result on my Ubuntu 16.04.5 LTS x86 based machine: the same error report
Note that again If I change the producer so that it uses a videotestsrc, there is no problem at all.

nasafix-nasser, 2019-02-19:
> Update.... So I tried using a website for an online webcam test: https://www.onlinemictest.com/webcam-test/. and the device there seems to be working fine. I get my webcam view, and the video goes away when I kill the producer instance. If that might help in finding the issue.

I checked that and the problem persists. I can see the first frame on the web site. This is just the same result that I get when I use my ARM based hardware.

mrdis, 2019-02-21:
I don't know if this may work for you, but in my case adding max_buffers=2 when I load the module can b used as a workaround.
`insmod v4l2loopback.ko max_buffers=2`

nasafix-nasser, 2019-02-26:
> I don't know if this may work for you, but in my case adding max_buffers=2 when I load the module can b used as a workaround.
> `insmod v4l2loopback.ko max_buffers=2`

Thank you it solved all the problems. 
But any body know why?

umlaeute, 2019-02-26:
there's a problem in the code  for queuing/dequeuing buffers, unless for the trivial case of 2 buffers.

probably this should be the default.

nasafix-nasser, 2019-02-27:
I found a special use-case in which I needs 3 buffers to work properly!

umlaeute, 2019-02-27:
well, i'm really talking about the *default* value which should work for most use-cases.
for special cases, you can always set the `max_buffers` to whatever you need.

pauldotknopf, 2020-06-09:
I'm still experiencing this issue, as of d45a49261f42ea7b04c821516dd2b81358bb97dd.

gerald-ftk, 2021-05-24:
> For my scenario, I need more than 2 buffers (a large pipeline).

Did you manage to resolve your issue? I'm having a similar situation and cannot seem to fix it even with `max_buffer=2`. Am running on a Jetson Nano though.

```
user@jetson:~$ gst-launch-1.0 v4l2src device=/dev/video1 ! xvimagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate a buffer
Additional debug info:
gstv4l2src.c(998): gst_v4l2src_create (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0
Execution ended after 0:00:00.316928037
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

```