v4l2loopback/v4l2loopback #167: Installation problem

lddd, 2018-03-23:
Hey there,

i am getting the following error on "make". Any idea how to fix this?

```
[root@CentOS-74-64-minimal v4l2loopback]# make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/root/v4l2loopback modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
arch/x86/Makefile:166: *** CONFIG_RETPOLINE=y, but not supported by the compiler. Toolchain update recommended..  Schluss.
make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.21.1.el7.x86_64'
make: *** [v4l2loopback.ko] Fehler 2
[root@CentOS-74-64-minimal v4l2loopback]# 
```

kripton, 2018-03-25:
Same as #163? Looks like you are trying to use a different compiler (version) as the one that was used to compile your kernel. Either update your toolchain (as suggested in the error message) or try building the whole kernel yourself, setting CONFIG_RETPOLINE to =n

umlaeute, 2018-03-25:
indeed, this looks very much like the same as #163 

@lddd please do not blindly repost your issues. if you have anything to add to #163 (like: not agreeing with the assertion that this is not a v4l2loopback problem), please add it there.

closed 2018-03-25 by umlaeute

lddd, 2018-03-26:
Thanks didn't notice about #163. Will try to update toolchain