v4l2loopback/v4l2loopback #173: How to Install this package in a target board ?

amsthecodder, 2018-04-21:
Hi,
I am a newbie to embedded linux and v4l2loopback.
I want to install v4l2loopback drive in my embedded linux.
I am using ptxdist to build my linux image file.
My linux distro is debian based.
I can not install v4l2loopback directly on a build as my board has no internet connection.
When I try with my build rule file, I am not able to compile this cause the v4l2loopback make file use uname for its kernal name.

So is there any other way to install this file?

umlaeute, 2018-04-21:
you can override the use of `uname` by specifying the `KERNELRELEASE` parameter to make.

to mimick the default behaviour, use:

    make KERNELRELEASE=`uname -r`

closed 2018-04-21 by umlaeute

amsthecodder, 2018-04-22:
Hey, thanks for the reply.
As you said I edited make file . 
make KERNELRELEASE=`uname -r`
My output after the changes is as follows: 

---------------------------------
target: v4l2loopback.extract.post
---------------------------------

patchin: fixup:
patchin: fixup: done

finished target v4l2loopback.extract.post

----------------------------
target: v4l2loopback.prepare
----------------------------

No prepare tool found. Do nothing.
finished target v4l2loopback.prepare

----------------------------
target: v4l2loopback.compile
----------------------------

make[1]: Entering directory `/home/myubuntu/linux/platform/target/v4l2loopback-master'
Building v4l2-loopback driver...
make -C /lib/modules//build M=/home/myubuntu/linux/platform/target/v4l2loopback-master 
make: Entering an unknown directory
make: Leaving an unknown directory
make: *** /lib/modules//build: No such file or directory.  Stop.
make[1]: *** [v4l2loopback.ko] Error 2
make[1]: Leaving directory `/home/myubuntu/linux/platform/target/v4l2loopback-master'
make: *** [/home/myubuntu/linux/platform/target/state/v4l2loopback.compile] Error 2

Is this  v4l2loopback app is only for runtime installation on the target?
As I informed you before I am not installing this app directly on target.
Using ptxdist I will create a linux image on my linux PC.
And then I will flash it to target.  

Do I have to do few more changes in the make file ?

Is there any other v4l2loop tar file is available which has configure script file?
I think this will make compiling and installing easy.

Or any other application using which I can create video loop.
I am going to use v4l2loopback for gstreamer as it is giving me an error /dev/video0 is not an output device.
 
Thank You.

umlaeute, 2018-04-22:
sorry, i'm pretty sure i don't really understand your answer.

- you do *not* need to edit the Makefile (at least, i'm pretty sure that you don't)

- running `make KERNELRELEASE=$(uname -r)` will build the module for your currently running kernel - just like a normal invocation of `make` without arguments; this means the kernel on the machine you are building on, not the target machine.
the reason why i posted it was to show you how to actually use the `KERNELRELEASE` option (without having to modify the Makefile) - and I don't  have the slightest idea of your target kernel. you should replace the `$(uname -r)` with an appropriate value for your target system.
you also might have to adjust the `KERNEL_DIR` variable, and possibly the `CC` variable (to select a suitable cross-compiler).

from your output i gather, that you most likely broke the build system by modifying the Makefile.

Finally `v4l2loopback` is not an application.
it is a *driver* for your kernel, that creates a kernel device.

anhthoai, 2019-07-13:
Hello.
I got this error and don't know how can handle it. Could you please help me?
make KERNELRELEASE=$(uname -r)
Building v4l2-loopback driver...
make -C /lib/modules/3.4.113-sun8i/build M=/root/v4l2loopback modules
make[1]: Entering directory '/lib/modules/3.4.113-sun8i/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/3.4.113-sun8i/build'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2

umlaeute, 2019-07-15:
@anhthoai please do not hijack other threads. for new problems, start new threads (I#m removing your post for now, and will remove *this* post  soonish)