v4l2loopback/v4l2loopback #176: having error trying install

toshibochan, 2018-05-05:
pi@toshiboPI3:~/v4l2loopback $ make && sudo make install
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback modules
make[1]: *** /lib/modules/4.9.35-v7+/build: No such file or directory.  Stop.
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2

please help

toshibochan, 2018-05-05:
this device is raspberry pi jessie

umlaeute, 2018-05-05:
please follow the "install" instructions in the README.

closed 2018-05-05 by umlaeute

umlaeute, 2018-05-05:
i just noticed that you've created multiple (duplicate and/or bogus) issues in the last few hours. please don't.
if you feel that i have closed an issue without justification, please add more information to the (closed) issue (as it can be opened again).