v4l2loopback/v4l2loopback #180: timecode passthrough

rogergodspeed, 2018-05-19:
Hi,

This is more of a question than an issue. I am working on extending some video for linux tools to make use of timecode as described in section 3.6.2 of the Video For Linux buffer documentation:

https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/buffer.html

I came across this very helpful project and was wondering if you would be open to extending support here for passthrough timecode as well. I haven't totally familiarized myself with the v4l2loopback code, but figure it doesn't make sense to pass timecode along if the producer doesn't provide it. That is to say v4l2loopback should not generate timecode in the same way that it makes up for missing timestamps.

What do you think?

umlaeute, 2018-05-23:
sure, go ahead.

however, it would be good if the timecode-passthrough could be disabled (per device).

rogergodspeed, 2018-05-23:
HI,

1) Is there an existing example of disabling any other feature per device that I should use as a model for timecode?

2) As I said I am still familiarizing myself with this project as well as video for linux. The documentation for the v4l2_buffer struct and timecode says:

"When type is V4L2_BUF_TYPE_VIDEO_CAPTURE and the V4L2_BUF_FLAG_TIMECODE flag is set in flags, this structure contains a frame timecode. [...] This field is independent of the timestamp and sequence fields."

Does that mean to you that only V4L2_BUF_TYPE_VIDEO_CAPTURE type buffers should ever contain valid timecode structs? In other words is there no way to loopback timecode in a round trip manner (from V4L2_BUF_TYPE_VIDEO_CAPTURE buffers to V4L2_BUF_TYPE_VIDEO_OUTPUT buffers)? Or have I totally misunderstood the buffer types?

rogergodspeed, 2018-05-23:
Hi,

I am using the following command with l4v2loopback managing /dev/video1 to create a fabricated capture device:

ffmpeg -re -f lavfi -i testsrc=size=1920x1080:rate=24 -flags -global-header -f v4l2 /dev/video1

It is my understanding (albeit still novice to this subject material) that ffmpeg is providing lv42loopback with V4L2_BUF_TYPE_VIDEO_OUTPUT buffers. I am then testing the result by running:

ffmpeg -f v4l2 -s 640x360 -i /dev/video1 -pix_fmt yuvj420p -c:v libx264 -f mp4 /tmp/foo.mp4

Where I believe v4l2loopback is providing ffmpeg with V4L2_BUF_TYPE_VIDEO_CAPTURE buffers. Is this accurate?

If my understanding is correct, then ffmpeg will have to insert timecode into the 'OUTPUT buffers and v4l2loopback will have to preserve those values by copying them into the 'CAPTURE buffers. This all seems to hinge on the idea that V4L2_BUF_TYPE_VIDEO_OUTPUT type v4l2_buffers with populated timecode structs are valid buffers, which is not my immediate -- however desired -- interpretation of the video for linux section 3.6.2 documentation.

If my reading of the documentation is correct, then I either need to hack my copies of ffmpeg and v4l2loopback to operate in the "broken" way I want, or I could instead pursue generating falsified timecodes from within v4l2loopback exclusively for the 'CAPTURE buffers. Neither path would be something I would expect or want to contribute to either project since that would be a hack.

Please tell me what your interpretation of the documentation is.

P.S. In my actual "production" use case I will be using the ntv2_v4l2 driver (which only deals in 'CAPTURE buffers) to transcode camera captures with ffmpeg. That case can rely on valid timecode structs in the 'CAPTURE buffers. Timecode support still needs to be added to both of those projects as well. My motivation to add support here was to fake the capture device for automated testing without the expensive setup and infrastructure needs of a real camera and capture card.

umlaeute, 2018-07-02:
hmm, in the description of the `timestamp` field in §3.6.2 I read:

> For output streams that use `V4L2_BUF_FLAG_TIMESTAMP_COPY` the application has to fill in the timestamp which will be copied by the driver to the capture stream.

which for me says that there are valid cases to have OUTPUT buffers with popuplated timecode structs (the  `V4L2_BUF_FLAG_TIMESTAMP_COPY` is esp. for mem-2-mem devices, like `v4l2loopback`)

rogergodspeed, 2018-07-02:
That makes sense. I think that would be an ok path in the future.

Sadly the combination of capture card and v4l2 driver I planned to use were too restrictive for my needs and I have instead written a collection of applications with the card's SDK. I wanted to leverage the rich v4l2 tool suite (like this awesome loopback driver), but when faced with rewriting the l4v2 driver for this card it was easier to make a libavdevice for ffmpeg directly.