v4l2loopback/v4l2loopback #183: Device not showing up in Chrome (sorry I know it's kind of a duplicate)

rivertam, 2018-05-31:
First of all, I'm sorry because I know this is kind of a duplicate of #78 and #40, but I didn't quite understand the solutions and I'm not sure if they're still relevant.

### Environment

  * `v4l2loopback` version: 0.9.1 (apt-show-versions says "0.9.1-4")
  * kernel version: Linux berman 4.13.0-43-generic #48~16.04.1-Ubuntu SMP Thu May 17 12:56:46 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
  * Distribution (+version): Ubuntu 16.04.3 LTS
  
### Problem Description

#### Steps to reproduce (with observations):

  1. `sudo aptitude install v4l2loopback-dkms`
  2. `sudo modprobe v4l2loopback devices=1 exclusive_caps=1 # I've also tried other, similar commands`
  3. `ffmpeg -f v4l2 -list_formats all -i /dev/video3 # and observe a message including "Not a video capture device." (expected)`
  4. `ffmpeg -re -i drop.avi -f v4l2 /dev/video3 # drop.avi is a test avi file and /dev/video3 is the correct device, I'm almost positive)`
  5. (while previous is running) `ffmpeg -f v4l2 -list_formats all -i /dev/video3 # and observe a message that is different, included below`
  6. (still while previous is running) Open Chrome console where `getUserMedia` has already been called and accepted. Run `(await navigator.mediaDevices.enumerateDevices()).map(a => a.label)`. Observe names (I'll just say, less mechanically, I recognize all of these and none of them are the loopback).
  7. Just to be certain, `cat /sys/devices/virtual/video4linux/video3/name` outputs "Loopback video device 0" which was most certainly not in the output of step 6.
  
Exact output from step 3:

```
ffmpeg version 2.8.14-0ubuntu0.16.04.1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.9) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
[video4linux2,v4l2 @ 0x13b7f60] Not a video capture device.
/dev/video3: No such device
```

Exact output from step 5:

```
ffmpeg version 2.8.14-0ubuntu0.16.04.1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.9) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
[video4linux2,v4l2 @ 0x1dbcf60] Raw       :     yuv410p :     Planar YUV 4:1:0 : 256x240
/dev/video3: Immediate exit requested
```

#### Expected Results:

I was happy to see that `ffmpeg` recognizes a valid device when and only when I have a producer. However, I was unhappy to see that Chrome is not recognizing the device. I have tried restarting my Chrome fully as well as incognito mode, which did change the devices available (I was surprised to see this) but did not show me the loopback device.

Again, I'm sorry if this is trodden ground, but I don't see this problem exactly anywhere else, and clearly it's not quite working for me (as I need the browser to be able to read from the stream).

---------

My fuller usecase (I doubt this is relevant, but might be good to note) is to try to create a producer in C++. At least at first, I'm going to try to stream individual frames into ffmpeg. I'm not sure exactly how I'm going to do this, so any advice about that is welcome, but that's a later step in my process. Basically, I'm trying to send still images I'm getting from a camera with poor device integration to the browser.

rivertam, 2018-06-01:
So I guess the `exclusive_caps` option only applies when you install from source? Not sure, exactly, but when I did `sudo modprobe -r v4l2loopback` and installed through the git repo using `make` and then `sudo insmod v4l2loopback.ko exclusive_caps video_nr=3 card_label="Fake"` it started working at least somewhat properly. (at least it shows up under devices. =) )

closed 2018-06-01 by rivertam

umlaeute, 2018-06-01:
I'm not sure what you mean with:
> I guess the exclusive_caps option only applies when you install from source

the `exclusive_caps` is a module parameter, which you can apply when loading the module (as you successfully did).

it has nothing to do with compiling from source (whatever that means; even when installing the `v4l2loopback-dkms` package of your distribution, you are in fact compiling the module from source; i don't think there are any binary packages of `v4l2loopback` in any linux distribution).

but probably some other bugs have been fixed, so using a newer version of the module helped you there...

rivertam, 2018-06-01:
> but probably some other bugs have been fixed, so using a newer version of the module helped you there...

Sorry, yes, that's more what I meant. Using the `exclusive_caps` option did not seem to work for me when I used `modprobe` with the aptitude package, where it did when I used the output of `make`.

rivertam, 2018-06-01:
I'm also realizing that perhaps my example drop.avi is too short -- it's about 15 seconds long. I'm getting somewhat inconsistent results when I use it, which might just mean Chrome is slow to pick up on sources. When I use a source of much longer (minutes, at least), I get much more consistent results.

ROBERT-MCDOWELL, 2018-09-26:
sorry to rebirth this thread but I'm facing to an issue with insmod and modprobe
modprobe load and unload v4l2loopback without problem,
but when I use modprobe the device is shown only in NON chrome/yandex/opera browser

using insmod is really strange, I used to successfully insert v4l2loopback only one time I don't know how
but after that I got

insmod v4l2loopback.ko exclusive_caps=1
ERROR: could not insert module v4l2loopback.ko: Unknown symbol in module

or

insmod v4l2loopback.ko exclusive_caps=1
ERROR: could not load module v4l2loopback.ko: No such file or directory

any idea?

umlaeute, 2018-09-26:
@ROBERT-MCDOWELL most likely the kernel has none of the v4l2l framework modules loaded.
If you did a `make install` you should also do a `depmod -a` (as detailed in the [README.md](https://github.com/umlaeute/v4l2loopback/blob/master/README.md)). (or load the `videodev` module manually)

ROBERT-MCDOWELL, 2018-09-26:
ok depmod -a was the beast.
now I can use insmod but browsers result is the same
seamonkey,firefox = ok
chrome,chromium = invisible
I tried
sudo insmod v4l2loopback.ko exclusive_caps=0 and 1 with same results
maybe it comes from a user security sandbox? I use root user for now.

ROBERT-MCDOWELL, 2018-09-26:
btw, you can test in real time in FLash here
https://www.onlinemictest.com/webcam-test-in-adobe-flash/
and in HTML5 here
https://webrtc.github.io/samples/src/content/devices/input-output/

ROBERT-MCDOWELL, 2018-09-27:
maybe is there anything related to driver signature restriction?
I know that edge, IE, but probably Chrome and chromium are checking the driver signature on windows. would it be the same on Linux too?
if it's the case I'm ready to contribute to pay the certificate. Other users should also contribute it

umlaeute, 2018-10-01:
using `exclusive_caps=1` works here with both chromium and firefox (both HTML5; i don't have flash installed and don't plan to).

I had to configure my audio to work with chromium first (else it wouldn't access the video).

This is with `v4l2loopback v0.12.0`

You might also want to try to set the `card_label` as indicated by @rivertam

ROBERT-MCDOWELL, 2018-10-01:
Ah ok, that's why, my tests are done with a computer without audio card.
Flash reacts exactly as HTML5 so there is no worries about it.

camwhite, 2018-10-08:
I'm facing this after upgrading to Ubuntu 18.04 LTS, while it worked on 16.04. Anyone have any ideas?

umlaeute, 2018-10-08:
@camwhite no.

- as I don't use Ubuntu myself, it would be *very* helpful if you could provide more information, notably which kernel and which v4l2loopback versions you are using (both in the working and non-working state); and which version of which browsers (don't) show the problem.

- as I use Debian/unstable, I'm usually ahead of whatever versions of Ubuntu. so the general functionality should be covered on my side (although I don't regularly test; so I might miss combinations of various versions)

- what have you tried so far? (`exclusive_caps`, `card_label?,...)

camwhite, 2018-10-08:
@umlaeute sorry that was not a constructive comment. Here is the relevant information..

## Working
```
Ubuntu 16.04.5 LTS (64 bit)
Kernel Name:  Linux
Kernel Version:  4.15.0-32-generic
v4l2loopback Version: 0.10.0
```

## Not working
```
Ubuntu 18.04.1 LTS (64 bit)
Kernel Name:  Linux
Kernel Version: 4.15.0-36-generic
v4l2loopback Version: 0.10.0
```

All browsers I have tested in show this problem.
- chrome
- brave
- firefox

I have tried loading the module with the above mentioned arguments without any noticeable effect. Interestingly, the dummy device is available to other programs like [OBS](https://obsproject.com/) and [guvcview](http://guvcview.sourceforge.net/), while it can not be detected by [cheese](https://wiki.gnome.org/Apps/Cheese) and the browsers. 

I have noticed that there is a newer version of v4l2l available, but not yet packaged. I will try building latest from source.

umlaeute, 2018-10-08:
yes, please try the latest version of `v4l2loopback`, which is **0.12.0**.

also, the new (0.12.0) is packaged just fine (but not in ubuntu-18.04; it's part of 18.10)

camwhite, 2018-10-08:
Removing and reinstalling `v4l2loopback-dkms@0.10.0` through apt has solved my issue. Thanks for your time @umlaeute !

tayyebi, 2020-04-24:
I have same problem with Ubuntu 18 (Zorin OS) and uninstalling didn't help me.

umlaeute, 2020-04-25:
"Ubuntu 18" is very ambiguous.
as i've mentioned in https://github.com/umlaeute/v4l2loopback/issues/183#issuecomment-427932912, Ubuntu-18.10 has a fixed version, but Ubuntu-18.04 does not..

dpocock, 2020-04-30:
I made some tests of 0.12.1 with both Firefox and Chromium, I've described my observations here:

https://bugzilla.mozilla.org/show_bug.cgi?id=1622628#c4

and I've published my gstreamer script, a basic example of a virtual webcam for chroma keying:

https://gitlab.com/freertc/gstreamer-chroma-key

Looking at comment 10 in the Google Chromium bug report, it appears they are unhappy about V4L2_CAP_VIDEO_M2M so I tried to comment out that line in the module source code but it still didn't work.  Somebody may need to look more closely at the Chromium / Chrome source code to decide how to modify v4l2loopback.  As I don't regularly use Chrome it isn't a big priority for me.

NLHybrid, 2020-05-14:
This guide worked perfectly for me. https://srcco.de/posts/using-obs-studio-with-v4l2-for-google-hangouts-meet.html

danilogit, 2020-06-03:
> This guide worked perfectly for me for getting it working google-chrome. It also, helped get some more consistency from obs too. Before I set it up this way, it was a mixed bag every time I tried to start up.
> https://srcco.de/posts/using-obs-studio-with-v4l2-for-google-hangouts-meet.html

Thank you! This have worked perfectly for me on ubuntu 18.04 ! 
```
sudo  modprobe -r v4l2loopback
sudo modprobe v4l2loopback devices=1 video_nr=10 card_label="OBS Cam" exclusive_caps=1
```